lucysrausch/hoverboard-firmware-hack #1: how to read serial with USB-UART adapter and arduino

chulo601, 2018-03-20:
Hi

I was able to flash the main board. after i flash the board, when i turn on the hoverboard the wheels start rotating but when i connect the USB-UART adapter to Gnd. PA2 & PA3 to get commands and i dont get any output on the serial monitor. what im doing wrong? i will like to control it with arduino. any help will be appreciate it

thank you

Tirta01, 2018-03-23:
Hi @chulo601 can you post your code? I using ppm branch and analog stick to ppm with arduino as controller, but sometimes the mainboard to off itself many times. ( link here: [https://github.com/kagrawal2/Drone-Transmitter](https://github.com/kagrawal2/Drone-Transmitter) )

I see this is very difficult for end users who do not understand the programming. please brother @NiklasFauth give a little examples of controllers code, so that many people can take advantage of your great work.

thank you

wholepair, 2018-03-23:
Good morning @chulo601 and @TirtaI - did you add UART control to the code?
I am using a ppm signal created from a RaspberryPi3 but would love to use
UART.
https://github.com/wholepair/USB-joystick-to-PPM-RPi3-Saitek.P990

Tirta01, 2018-03-24:
> I am using a ppm signal created from a RaspberryPi3 but would love to use
UART.
https://github.com/wholepair/USB-joystick-to-PPM-RPi3-Saitek.P990

Yes i tried uart control but can not make it work. I asked at arduino forum for uart control [http://forum.arduino.cc/index.php?topic=531195.0](http://forum.arduino.cc/index.php?topic=531195.0)

@wholepair I see your video thats works great, is there any issue with the mainboard sometimes it's hard to turn on with PPM-RPi3-Saitek.P990? 

my test with analog to ppm with arduino nano.
[https://www.youtube.com/watch?v=Cp56o8AWFB0](https://www.youtube.com/watch?v=Cp56o8AWFB0)

chulo601, 2018-03-24:
@Tirta01 @wholepair thank for your help. i was using https://github.com/OpHaCo/hoverbot to try to read the serial data but im not getting any serial data. did you guys edit the main.c to get the PPM signal from the serial data?

TomTinkering, 2018-04-16:
As far as I can tell, UART is only setup to send data from the board over the serial-port for debugging. 

The control methods available right now are:
  - Wii Nunchuck
  - ADC inputs
  - PPM

These are selectable in the code, but need re-compiling. If you want to add UART control, you will need to setup one of the UARTS to also receive data, and use that data to set the pwml/pwr variables. This will require some knowledge of the code, but it is a good project to learn.

I am working on a project for which I have built a UART interface, but it also includes some drastic other changes in the code. Can I ask what your intended goal is? Maybe I can help you with some snippets.

TomTinkering, 2018-04-16:
Have a look at 
https://github.com/NiklasFauth/hoverboard-firmware-hack/blob/master/Src/control.c

I haven't used this myself, but it seems the I2C port 2 is used (it is available through one of the sensor-board connectors)

wholepair, 2018-04-16:
> control.c

It looks like it is the default control method too.
https://github.com/NiklasFauth/hoverboard-firmware-hack/blob/master/Inc/config.h

I am going to test this code later today.  I want to make sure I connect my
nun-chuck correctly.

I am assuming that the connection is 'not' that same as the one here:
https://www.hackster.io/salviador/wheelchair-hack-with-hoverboard-8067bc

TomTinkering, 2018-04-16:
It is the same :) The code is different, but there is only on I2C port available without having to solder directly to the MCU pins, that is the same one as used by Salviador.

wholepair, 2018-04-16:
I think I have got it now.

SCL2
SDA2
+5v
gnd

I am excited! I was working with this a bunch last month but I took a break
from it for a few weeks.  Time to dive back in!

TomTinkering, 2018-04-16:
Sounds good :) Seems like you are right about the default method.

Right now the code is set to listen to the Wii-Nunchuck, and output DEBUG info on UART2 (the other sensor-board connector). I'm assuming this corresponds with the Hex file in the repo. You might have to rebuild it though.

NiklasFauth, 2018-04-16:
Hey,

thanks for answering some of the questions. For some reason I had github notifications disabled and just found this issue. 

I'm working on this project to fit my demands of controlling the motors, so I add new features as I need them. As TomTinkering said, UART is currently supported in TX mode only. However, the previous version of this firmware supported receiving floats via UART DMA to set the motor speed accordingly. Have a look at https://github.com/NiklasFauth/Hoverboard-Board-Hack/blob/master/Src/uart.c and https://github.com/NiklasFauth/Hoverboard-Board-Hack/blob/4a0ecf17a2fd2ca237823c69aa00e6476852ba7b/Src/telemetry.c#L22 if you need a starting point for porting this feature. I would not recommend to use this old firmware as it is tho, as it has some basic problems.

If you want to connect a nunchuck, the current config in this repo should work out of the box. Adding the nunchuck was a quick-and-dirty evening project and I want to read data via DMA in the future, but for now it works. 

Connecting it is quite straight forward, don't forget pullups and a voltage regulator. The voltage from the sensor board connectors is 15V, so you will need a AMS117-3.3 or similar to get 3.3V for the nunchuck and pullups. 

A friend of mine did a nice breakout to connect a Nunchuck, UART, and analog potentiometers:

https://github.com/Jan--Henrik/hoverboard-breakout

Also we will have a talk at the GPN at May 10-13 (https://entropia.de/GPN18/en) that will be live-streamed, and we will try to make it as a basic beginners guide to getting started with hoverboard hacking.

wholepair, 2018-04-16:
I am all setup to rebuild and flash with a black magic probe -
https://1bitsquared.com/products/black-magic-probe

chulo601, 2018-04-16:
@TomTinkering  @NiklasFauth thanks for your reply. I got a great deal on about 100 used & broke hoverboard and i want to start used the brushless motors but esc are expensive so i want to use the main board to start building robotic arm, four wheels skate board and robots. i will like to use arduino code to control it until i can more advance with the coding. What IDE you use to recompiling the code? @TomTinkering @NiklasFauth i have extra main board i can send you if you need them for testing, shipping will be cheaper for the main board because the motors are to weights a lot.

TomTinkering, 2018-04-17:
Hey Chulo, perhaps move this discussion to another place as it is not related to the original issue.
(forum, new issue, email?) As for the IDE, I personally use STM32-Workbench with an ST-Link V2 debugger. Works great, but it requires a bit of setup.

closed 2018-05-07 by NiklasFauth

alex-makarov, 2019-07-26:
@NiklasFauth Sorry for plugging into an old thread, but I cannot find details on connecting the UART to, for example, a 3.3v Arduino. What is the signal level there, do I need a  AMS117-3.3 for RX and TX? Thanks!

p-h-a-i-l, 2019-07-27:
@alex-makarov your chances of finding answers are better when you open a clean new issue, which is also open then.

Anyways, singla level of the STM32 is 3.3V, so you can directly attach it to your arduino if it operates indeed on 3.3V.