lucysrausch/hoverboard-firmware-hack #2: Infinite Loop

Candas1, 2018-04-16:
Hi

Thanks a lot for having shared this code.
I flashed it to my hoverboard but the code is stuck in Default handler/Infinite_Loop after one loop.
So wheel rotate a little bit, then nothing.

Have you ever experienced that issue?

Thanks in advance.

Candas

Candas1, 2018-04-16:
Hi Tom,

I just compiled the code as I needed to use control through PPM and wasn't sure for which control type the hex file was compiled.

In the meantime I tried PPM branch from old repository and it worked.
Thanks for the hint, I will investigate that.

NiklasFauth, 2018-05-07:
Something that just came to my mind that might be the problem: After flashing, you need to disconnect the jumper on the power button. When turning on the Mainboard with a button press, the firmware stays in a while loop until the button is released. If you keep pressing the button or you have a permanent jumper on this connector, the firmware won't boot.

closed 2018-05-14 by NiklasFauth