lucysrausch/hoverboard-firmware-hack #4: unable to unlock

k3nden, 2018-04-27:
Your work is very exciting and thank you for sharing. This my first effort at trying to flash an off-board chip so there is a lot of learning for me.  I am trying to unlock the chip with the README openocd command and receive the following
Info : device id = 0x13090414
Warn : STM32 flash size failed, probe inaccurate - assuming 512k flash
Info : flash size = 512kbytes
Info : Device Security Bit Set
Error: timed out while waiting for target halted
target state: halted
target halted due to debug-request, current mode: Handler HardFault
xPSR: 0x00000003 pc: 00000000 msp: 0xffffffe0
Error: error waiting for target flash write algorithm
stm32x failed to lock device

I am using a st-linkV2 programmer, ubuntu 1604. I have tried to find help for this problem but nothing seems to help. I would appreciate knowing more about your development environment so that I might a a reference point as to what might be the problem. Thanks.

HobbyIngenieur, 2018-05-01:
Same here. OpenOCD produced some timeouts but didn't unlock. Using the ST Link Software and just setting the "Read Out Protection" in the Option Bytes menu to disabled works fine. When applying the changes, the ST Link Software states, that it wasnt able to write the Option Bytes to the Chip, but just don't care :) Afterwards the chip can be flashed :)

Thank you very much for your work!

NiklasFauth, 2018-05-07:
Hey,
well, for me I had some experience with STM32 before, but I don't think it is really necessary. Especially because all the hard hardware initialization is already done, so all that's left to do for you is basic C / C++ programming which is the same for Arduino.

I've heard about the problem of unlocking the STM32 from different people now, and also experienced the difficulties myself. The thing is, for original STM32 the command in the Makefile works fine. However, most Hoverboards are assembled with counterfeit STM32 from GigaDevice, and they are mostly, but not exactly compatible. You might need to reset additional registers to get around the readout protection. Here are some commands that worked for mainboards that couldn't be unlocked before:

`openocd -f interface/stlink-v2.cfg -f target/stm32f1x.cfg -c init -c "reset halt" -c "mww 0x40022004 0x45670123" -c "mww 0x40022004 0xCDEF89AB" -c "mww 0x40022008 0x45670123" -c "mww 0x40022008 0xCDEF89AB" -c "mww 0x40022010 0x220" -c "mww 0x40022010 0x260" -c "sleep 100" -c "mww 0x40022010 0x230" -c "mwh 0x1ffff800 0x5AA5" -c "sleep 1000" -c "mww 0x40022010 0x2220" -c "sleep 100" -c "mdw 0x40022010" -c "mdw 0x4002201c" -c "mdw 0x1ffff800" -c targets -c "halt" -c "stm32f1x unlock 0"`

`openocd -f interface/stlink-v2.cfg -f target/stm32f1x.cfg -c init -c "reset halt" -c "mww 0x40022004 0x45670123" -c "mww 0x40022004 0xCDEF89AB" -c "mww 0x40022008 0x45670123" -c "mww 0x40022008 0xCDEF89AB"  -c targets -c "halt" -c "stm32f1x unlock 0"`

Hope that helps!

Mr0Inka, 2018-05-07:
Wow. That was fast! I am trying to control a (eventually single) hoverboard wheel with a potentiometer. Would need this for a prototype (industrial design student from stuttgart here, hi :P).

I'm actually not that far yet, so I can't really tell if I am able to unlock the STM or not. I've been trying to understand how everything around compiling and flashing for STM32 works, but well. Obviously more complicate than Arduinos.  

Would you recommend the STM32 System Workbench? I am already having troubles getting openocd installed and working. Not even talking about me not knowing what exactly a "gcc-arm-none-eabi binary" is or which files exactly I am supposed to use. I heard that it's actually easier on linux.

Been trying different approaches i found on Github for like 14 days now and even bought a F446RE because the "OpenChair" project used that one to immitate the sensor inputs. No luck tho.

My main question was, if you'd say, that this can be done without having much knowledge about STM32 in general and all the tools around it or if I should just leave this and try a different way.  

Thanks a lot for all the hard work you are putting into this!

Mr0Inka, 2018-05-07:
I think compiling worked now. Now i have a hover.bin and a ton of other files. Are those other files normal?  

And are those errors or just info logs? 

https://pastebin.com/E3Wh6d84

NiklasFauth, 2018-05-07:
Hey,

if you're from Stuttgart, you might want to travel to Karlsruhe next Friday. At the GPN (https://entropia.de/GPN18) I'll have a  talk about Hoverboard hacking (https://pretalx.entropia.de/gpn18/talk/RVCXHF/) and there is a workshop: https://entropia.de/GPN18:Moving_Objects

It's for free and if your into electronics it'll be a hell lot of fun :)
Also I could help you personally. If you don't have time on Friday, I'll also be there Saturday and Sunday.

And yes, you successfully build a working firmware. hoverNew.bin is all you really need (hoverNew.elf if you need debugging).

Mr0Inka, 2018-05-07:
That sounds good. Do I need to register? That workshop also sounds nice, but seems to be sold out. I have quite a few of the components tho (Board, mainboard, battery, a F446RE).

NiklasFauth, 2018-05-07:
You don't need to register for the GPN nor the talk. If you want to join the workshop, please add yourself to the bottom of the table on the wiki page. You can join the workshop with your own hardware. Only the amount of kits I have available is limited to 20.

Mr0Inka, 2018-05-07:
Okay, seems like I'm in. I only have a 350W Motor, so I'll take a set from you (I counted 18 sets so far on your list).

Also, the lecture is on 2018-05-11, 13:00–14:00 - but when is the workshop? I can't find a date/time here.

Mr0Inka, 2018-05-10:
Hey, I installed everything for the workshop on Ubuntu. but the toolchain seems to be v6? Is that okay or do i need to downgrade to 4/5 as told on the workshop page?  

`gcc-arm-none-eabi is already the newest version (15:6.3.1+svn253039-1build1)`

k3nden, 2018-05-11:
Have fun this weekend. I wish I could be there.  I still have not been able to make the Wii controller work.  when I power on one drive wheel starts and runs at a pretty high speed.  I am powering the wii  controller with an external battery.  ~5k pullups on the clk and data lines.  Should the motor run at startup? Any suggestions on how to trouble shoot the contoller?

wholepair, 2018-05-11:
Will any of the workshops be web streaming?

wholepair, 2018-05-11:
Found it!
https://media.ccc.de/v/gpn18-95-howto-moving-objects

rokrodic, 2018-05-30:
Hi. Niklas, the second reset method above work on my PCB. You should edit readme... Thanks!

closed 2018-06-11 by NiklasFauth

ACG-TR, 2020-03-25:
> Hey,
> well, for me I had some experience with STM32 before, but I don't think it is really necessary. Especially because all the hard hardware initialization is already done, so all that's left to do [...]

Hello I tried these lines but get below notes; can you help me out what to do? I cannot connect to the mainboard via ST  link, i gives an error code (cannot connect to target)

Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.      Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD   adapter speed: 1000 kHz                                                                                                 adapter_nsrst_delay: 100                                                                                                none separate                                                                                                           Info : Unable to match requested speed 1000 kHz, using 950 kHz                                                          Info : Unable to match requested speed 1000 kHz, using 950 kHz                                                          Info : clock speed 950 kHz                                                                                              Error: open failed                                                                                                      in procedure 'init'                                                                                                     in procedure 'ocd_bouncer'