lucysrausch/hoverboard-firmware-hack #7: Schematic and FOC motor control??

casainho, 2018-05-11:
Hi.

Can you guys point me to a schematic in case anyone already go it??

Can the hardware do FOC? For what I could understand from your code, there are 2 phase currents for each motor, right?

Does the motors run as much silent with your firmware as with original firmware?

Mr0Inka, 2018-05-12:
About silence: I can't tell a difference between stock and custom firmware.

TomTinkering, 2018-05-13:
Have a look at the schematics here:
http://vocke.tv/lib/exe/fetch.php?media=20150722_hoverboard_sch.pdf

Also salviador has one:
https://github.com/salviador/Hoverboard-Board-Hack/blob/master/board%20SCH.pdf

Be careful with the naming, it might be different from the code

TomTinkering, 2018-05-13:
As for FOC, I am not an expert, but I think you need at least 2 phase voltages per motor to do it. There are some voltage measurement circuits on the board, but I think they do zero-crossing detection (I might be wrong), have a look at p4 of the first schematic. You can probably change this circuit to do it, but as far as I know there is no software implementation for FOC on this board yet (or sensorless for that matter)

TomTinkering, 2018-05-13:
@NiklasFauth maybe its a good idea to add salviadors or my schematic to the repo? I saw that you have a nice overview on the readme page already for the connectors, but for some people who like to work on hardware it might be useful

casainho, 2018-05-13:
I would like to see a schematic -- if you have a link to it, please share.

"As for FOC, I am not an expert, but I think you need at least 2 phase
voltages per motor to do it. There are some voltage measurement circuits on
the board, but I think they do zero-crossing detection (I might be wrong),
have a look at p4 of the first schematic. You can probably change this
circuit to do it, but as far as I know there is no software implementation
for FOC on this board yet (or sensorless for that matter)"

It needs only battery voltage reading and 2 phases currents. There is no
need for sensorless since it has hall sensors signals from the motor.

Main advantage of FOC principle is to get most torque per amp, most range
possible from the battery.

I did FOC for STM32F103, for electric unicycles and I am doing now for
EBikes:

- https://eggelectricunicycle.bitbucket.io

- https://opensourceebikefirmware.bitbucket.io

à(s) 11:00:

TomTinkering, 2018-05-13:
A, phase currents, makes sense. There is only 1 DC current measurement per motor unfortunately.
See the link in my first comment for the schematic

casainho, 2018-05-13:
Missed the link before. On the schematic I see PH_AMP_A_OUT and  PH_AMP_B_OUT
-- aren't they phase A and B currents??

à(s) 12:44:

casainho, 2018-05-13:
By the way, see here my firmware for the same motor, first running with 6
steps/block commutation and after with SVM -- see the big difference in
noise!!

1. https://www.youtube.com/watch?v=O9shbfQ9hHk

2. https://www.youtube.com/watch?v=EonFwTk2cs8

Casainho <casainho@gmail.com> escreveu no dia domingo, 13/05/2018 à(s)
23:25:

> Missed the link before. On the schematic I see PH_AMP_A_OUT and  PH_AMP_B_OUT
> -- aren't they phase A and B currents??
>
> à(s) 12:44:
>
>> A, phase currents, makes sense. There is only 1 DC current measurement
>> per motor unfortunately.
>> See the link in my first comment for the schematic
>>
>>
>
>
> Cumprimentos,
> Jorge Pinto
> E-mail: casainho@gmail.com
> Telemóvel: 927145983

TomTinkering, 2018-05-14:
If you have a good idea to improve this than you are more than welcome to branch and work on it. If your interest is mainly the noise, I think a sensorless alghorithm based on the phase voltages above certain speeds will also reduce the noise? Or you can maybe add current transformers on your phases?

casainho, 2018-05-14:
I can't help on developing firmware as I am not motivated to work on
hoverboards. I had one and both me, little son and wife we quick learn
riding it but I also sold it fast then moved to electric unicycles, for
which I started to develop the firmware (EUC boards also use STM32F103!!).
Then I moved to EBikes, where I am now.

A  good idea is for you guys look with an oscilloscope and save pictures of
the PWM scheme used is, maybe it is block commutation only but with the
sync with BEMF using that zero crossing as you say. That sync would be the
same as FOC: field oriented control, which means put generated phase
voltages oriented (sync) to magnet field of the rotor magnets (reference
the BEMF).

The thing is that the Chinese engineers to a great work at keep a balance
of using FOC principles VS low price of the hardware. I would say you guys
aren't yet with same quality as original firmware -- if you guys are not
using that piece of hardware for zero cross detecting, that Chinese
engineers would not put on the board to not be used!!

But the importance of FOC is to get the best torque possible from the motor
and hence also the best battery range. I think if you guys can, you should
do that.

And I would like to help in this way: I do my projects as OpenSource and I
keep my notes public, and here is:
- example of original PWM:
-- 
https://opensourceebikefirmware.bitbucket.io/development/Motor_controllers--BMSBattery_S_series--BMSBattery_S06S--PWM_signals.html
-- 
https://opensourceebikefirmware.bitbucket.io/development/Motor_controllers--BMSBattery_S_series--BMSBattery_S06S--PWM_signals--very_low_speed_-_6_steps.html
-- 
https://opensourceebikefirmware.bitbucket.io/development/Motor_controllers--BMSBattery_S_series--BMSBattery_S06S--PWM_signals--low_speed_up_to_max_speed_-_sineware.html

-- 
https://opensourceebikefirmware.bitbucket.io/development_tsdz2/About_Tongsheng_TSDZ2_mid_drive_motors--Motor_controller.html
(see PWM signals)

So, I am being developing and sharing so others can take advantage. I also
take a lot of advantages from other OpenSource projects and I strong
recommend you guys to read great Shane Colton documentation and firmware
for STM32F103:

-
https://opensourceebikefirmware.bitbucket.io/development/Motor_control--FOC--Shane_Colton_documentation_and_firmware.html
-
https://eggelectricunicycle.bitbucket.io/FOC--Shane_Colton_documentation_and_firmware--Firmware.html
(firmware)

Respect for your work, you guys are doing something unique and special!!
Please keep it up!!

à(s) 09:13:

NiklasFauth, 2018-05-14:
From what I now, the original hoverboard firmware doesn't use FOC. However there are tons of slightly different firmware out there so some might actually do FOC. In principle, the hardware is capable of FOC, as the phase current of two phases per motor can be measured via the RDSon of two low-side mosfets.
I tried using this measurement not for FOC, but for an actual Phase current PID and it didn't work out so well. The measurements are horrible because the RDSon is highly nonlinear and the OpAmps are dirt-cheap. For the PID, I didn't continue testing because current chopping with DC measurement works good enough. IMHO, FOC would be very nice but definitely not my highest priority. If you want to do some testing, feel free to contribute to this project :)

@TomTinkering I will link the schematics in the ReadMe.

casainho, 2018-05-14:
Good to know that phase current can be measured using RDSon. I never did
that but I know that cheap EUC boards controllers do that and I alsosaw
some EBike motor controllers doing that also.

I would like to share that on my current firmware for Ebike motor
controllers, the microcontroller is STM8 (8 bits 16 MHz) but still does
some kind of FOC using 1 phase current sensor only, which I call very low
resolution FOC because it is done once every ERP and not every PWM cycle.

Also, the hardware has 2 signals from the shunt current sensor: battery
current and total phase current. Battery current is shunt signal but low
pass filtered. With battery current I do the motor current control and with
total phase current, I just limit it to avoid burning mosfest.

On other new Ebike motor controller, using also STM8, they do SVM just
like  on FOC but there is no phase current. I plan to try implement "FOC"
and I am pretty sure original firmware does, using calculation of the angle
and that needs inputs of motor speed, motor phase current and motor R and
L. As this motor controller only works for this motor, I think it can be
done like that.

So, that is just ideas and who knows if can help you guys in the future.

à(s) 12:43:

TomTinkering, 2018-05-14:
Ah nice, I didnt think of Rds on, interesting, I might look into this, would be a good learning project.

kloppertje, 2018-05-17:
Also interested in some more advanced motor controllers. I recently switched from original hoverboard FW to your modified FW, and noticed two things. The motors become slightly more noisy, and they have less torque at low speed. So, my gut feeling is that the original FW has a more advanced control than the modified one. Any ideas what it could be?

casainho, 2018-05-17:
That symptoms are, from my experience, because FOC control mode is not
being used.
The symptoms should be worst with higher currents and/or higher speeds.

But there could be a reason for you feel less torque as lower speeds, like
at higher speeds the angle is near the correct one just because some luck
on the implementation. Sometimes things work because we have luck.

On Ebikes we want to have the best torque possible. Also, other way to
compensate lack of torque is to increase the current but that will reduce
range. So we try to make best motor control possible.
And yes, everyone like most silent motor possible also.

NiklasFauth, 2018-05-17:
1. FOC does not necessarily require phase current measurement. You can do FOC without any measurement at all. All what FOC needs is that you interpolate the motor position between the known hall states by using an observer to get higher rotary resolution than 6 per pole, which enables more advanced control modes like using direct-quadrature-zero transformation (Clarke-transformation) or even space vector modulation (SVM).

2. however, for our application this is not necessary. We don't need FOC or SVM for a moving armchair or a bobbycar, because we don't need accurate regulation or advanced control. The most intuitive way to control something you are sitting on is by torque thus by motor current, and that works well enough with block commutation.

3. the reason you experience this difference between the hoverboard firmware and this firmware is that the hoverboard firmware uses actual controller to balance the user. Especially, when I reverse engineered parts of the original firmware I found online, I noticed that they use a cascade control (which totally makes sense for this application). They use a motor current PID, on top of that a velocity observer, and on top of that that actual position / angle control. By having an actual velocity PID you can have whatever torque even at low speeds, and that's why the hoverboard may seem to act different.
If you want the same behavior, go measure the motor speed (eg in the bldc.c function by observing the change rate of posr or posl), but a PID on top and you will have the exact same behavior. We did this. There is even some code left on earlier commit. https://github.com/NiklasFauth/hoverboard-firmware-hack/blob/47f243b2fe304a6ecaacea20e3c7e72f3711b7f2/Src/bldc.c#L228

However, you don't want this for moving objects. It's nice for robotics, or other control applications, not for applications where you control things by hand. It feels very messed up and I can guaranty you if you try it yourself, you will be very unsatisfied. 

Hope I cleared things a bit up. Please be very careful when you use magic buzzwords like FOC or SVM, as they are heavily misused recently by the copter community recently. Read the wikipedia article for what you are talking about and check if that really is what you want to talk about.

kloppertje, 2018-05-17:
Thanks for the very clear explanation! My intention is to use the hoverboard hardware for robotics, quite like the original purpose. However, I don't want a human on the board, but simply a mass of a few kg, such that the robot can self balance. In that case, somewhat more accurate regulation might give improved preformance, although the current implementation will probably work as well.

HITMAnsOFT, 2018-06-22:
@kloppertje There is another firmware project you can try that implements sinusoidal control and seems speed-controlled: [isabellesimova/HoverboardFirmware](https://github.com/isabellesimova/HoverboardFirmware). However I'm kinda reluctant to try it as I can't find motor current limiting or any other safety features in the code.

kloppertje, 2018-06-22:
@HITMAnsOFT looks nice :). I've also been looking into the ST Motor Control SDK (5.1 to be specific), which provides FOC "out of the box". I got it working with the standard evaluation board. Haven't tried it at low speeds, which is of course the most interesting part, as I didn't have a suitable encoder lying around. Don't know how much work it would be to implement code generated with the SDK for the hoverboard, quite some I'd expect.

casainho, 2018-06-22:
Still no FOC implemented. To improve, just look at Shane Colton good
tecnhical notes and firmware sources or at my code.

HITMAnsOFT, 2018-06-22:
@casainho Right no FOC, but still should be smoother and quieter.

casainho, 2018-06-22:
Sure it is and I have videos of the same motor running 6 steps/block
commutation and SVM, the difference is big. But after what you done, you
just need to add on top the angle displacement that will be an offset index
for the table values.

For KT motor controllers, I didn't calc the angle but did interactively up
to get the phase current synchronized with hall sensors signals (rotor
magnetic Flux / BEMF).

For TSDZ2 motor controller, I calc the angle using phase voltage ( battery
voltage * duty-cycle)  and I*w*L. I is motor ohase current (shunt current /
duty-cycle) and L is measured from the motor coils.

Jan--Henrik, 2018-10-10:
Hi @p-h-a-i-l , no sorry I don't know where to find the KiCAD files. But as @TomTinkering created them, you may want to ask him :)

closed 2018-11-10 by NiklasFauth