lucysrausch/hoverboard-firmware-hack #9: Motor disabled when braking

Mr0Inka, 2018-05-15:
Hey,  
I am using a single motor on a scooter, currently controlled by the wii nunchuck.  

- With no weight on the vehicle everything works fine. 

- If I am driving with atleast 50% speed, I can sowly decrease the speed (slowly moving the joystick back to the neutral position), which will successfully brake down and stop the vehicle. But ... if I brake to fast, the motor will just shut down and I'll have to restart the mainboard.  

Is there any way around this? Is it this line maybe? 

`#define TIMEOUT          5           //number of wrong / missing commands before emergency off`

NiklasFauth, 2018-05-16:
Hey,
the problem might be linked to the motor current that flows at faster breaking.
We experienced that especially cheap China Nunchucks loose connection when there are too much interference. Try using a original Nunchuck, better noise rejection (like a ferrite on the cable for common mode rejection) or add additional 100nF ceramic capacitors to the 3.3V rail inside your Nunchuck.

Mr0Inka, 2018-05-16:
I was also thinking some battery protection circuit being the problem (current flowing when braking). Do you think using a second wheel would improve this? Or would the flowing current from both wheel just "add up" and lead to the same result?

NiklasFauth, 2018-05-17:
When the STM32 can't establish a connection via I2C, the timeout variable is increased by one.
(https://github.com/NiklasFauth/hoverboard-firmware-hack/blob/9ef7e0b93655cf4690c53ca04196782c3e7e32e8/Src/control.c#L83)
After three communication problems, the STM32 tries to reinitialize the Nunchuck. If this doesn't help and the timeout variable exceeds the limit defined in the config.h, the motors are turned off for safety reasons:
https://github.com/NiklasFauth/hoverboard-firmware-hack/blob/9ef7e0b93655cf4690c53ca04196782c3e7e32e8/Src/bldc.c#L181

Mr0Inka, 2018-05-17:
Is the initialization one-sided from the STM or does the Nunchuk respond to something like an initialization request?   

And can the safety feature be disabled by setting the timeout to 0 for example? Or eventually enable the motor again after X seconds?

Mr0Inka, 2018-05-17:
The second motor improved braking by a lot. Shutdown only happen when braking really hard. I increased the AMP limit to 17 additionally. No idea how far I could go with that without barbecuing something.

NiklasFauth, 2018-05-18:
We tested the hardware up to 35A without problems. The 15A was chosen for the workshop to not break someones arm or whatever. Everything <30A should be safe.

Mr0Inka, 2018-05-18:
Awesome! Case closed then! thanks. Will try to set it to like 25A. That should be enough. :)

closed 2018-05-18 by NiklasFauth

p-h-a-i-l, 2018-07-27:
Hey
I'm experiencing lots of Issues with lost connections to the nunchuck, also during braking.
The reinitializing attempt does not seem to work at all - I experimented also with power cycling the nunchuck to check if it needed a reboot.
Next thing would be to add TVS diodes an capacitors to the nunchuck, but I think it would be nice to have the reinitialization code operational too.
(i'm using original nunchucks)