lucysrausch/hoverboard-firmware-hack #13: Firmware flash

DSBRobotics, 2018-05-17:
First Great job with project and video is awesome. 
My issue is putting firmware together and flashing, anyway you can detail that process out for an old newbie?
Also I have two hover boards that only use two boards so have you seen those and do you have a hack for those?

Thanks so much

LeoDJ, 2018-05-23:
Currently, the hoverboards with the split mainboard are not supported.
They would most likely need at least firmware tweaks and we don't have one to test it

As for flashing the firmware, maybe in the future we will provide a step-by-step instruction, but for now perhaps the [talk](https://media.ccc.de/v/gpn18-95-howto-moving-objects) will help you a bit.

flo199213, 2018-05-25:
@DSBRobotics A friend of mine and me, we are currently working on a firmware for the Gen2 split-mainboard Hoverboard version. We are having some problems with the new controllers. They are using a GD32F130 instead of the STM32F103 with a different mapping of the HAL-Registers and a different peripheral using. So we have to start from scratch (We tried to use the current firmware-hack, believe me :-)). Secondly we have to split our firmware into a master and a slave firmware because the two motor drivers are communicating via USART. Third challenge is that there are no free pins or connectors left to add a potentiometer, remote control or nunchuk. So you definetly have to solder, to add a steering device.
Currently my Motor Bridge is broken. I will keep you up to date via "Issue", if we have some new information on the new firmware working on our boards. Especially pseudo schematics which pin connects to wich peripheral etc, if I will find some time to draw them...

TonyBoyd66, 2018-09-24:
A quick guide for running the makefile and flashing an STM32 uC:
- Install [CYGWIN](http://www.cygwin.com/): allows running "make" commands in windows, with a linux-like shell.
!important! Install full content of "devel" package during installation to allow "make" commands to be run (takes a while to install)
- Install GNU ARM embedded [Toolchain](https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads): Allows to run commands written in the makefile which are ARM specific. 32bit version available only.
- Open Cygwin terminal and type cd /cygdrive/c/.../makefile_folder/
- Type "make" - You might have to change the arm-none-eabi... prefix in the makefile (i didn't change the default installation paths during the installation of the above software, so I didn't have to) - see github how-to.
- Hex content for STM32 is now in build folder
- Use STlink [utility ](https://www.st.com/content/st_com/en/products/development-tools/software-development-tools/stm32-software-development-tools/stm32-programmers/stsw-link004.html)software to interface with st-link v2 and flash
- Also, if like me you're using the official ST-Link v2 ISOL, know that the pinout is wrong on the datasheet. See thread [here](https://community.st.com/s/question/0D50X00009XkWYnSAN/problem-connecting-stlinkv2-isol-to-stlink-utility). Same goes with its latest firmware which doesn't necessarily allow accessing flash under reset. [Downgrading ](https://www.carminenoviello.com/2016/02/26/restore-st-link-interface-bad-update-2-26-15-firmware/) it is needed.

Good luck!

closed 2018-11-10 by NiklasFauth

gnbl, 2019-09-27:
Here's @flo199213's repository: https://github.com/flo199213/Hoverboard-Firmware-Hack-Gen2