lucysrausch/hoverboard-firmware-hack #16: Dont get the nunchuk control working

ctucx, 2018-05-20:
hello,
i dont get the nunchuk control working, i tried the the ADC control and that one is working.
i have tried both connectors, on two boards. nothing, and of cause i tried the nunchuk on an arduino: it works. i connected it to the board with 3k3 pullup on scl and sda...

Can someone help/ give tips?

NiklasFauth, 2018-05-20:
How do you power the Nunchuck when connected to the mainboard?

ctucx, 2018-05-21:
via the 3.3v rail.

Manawyrm, 2018-05-21:
Hi ctucx,

that's probably EMI. You could try shielding the cable (if you can). A capacitor against ground might also help.
Are you using a real nunchuck? I had trouble with the cheap chinese nunchuck clones.

ctucx, 2018-05-21:
yes, its a china clone. then i will buy a wireless nunchuck...

Mr0Inka, 2018-05-25:
For me the China nunchuk was super buggy. Imitating it with an Arduino worked without any problem and went super smooth tho.

rene-dev, 2018-05-25:
@NiklasFauth said they are missing bypass caps, adding 100n to the nunchuck fixed some issues.

closed 2018-08-09 by Jan--Henrik