lucysrausch/hoverboard-firmware-hack #18: Wheels getting "stuck"? Board shutting down

Mr0Inka, 2018-05-24:
Hey,  
it happened a few times, that the wheel suddenly stopped while going full speed. This doesnt feel like the slowing down when releasing a nunchuk joystick to neutral, more like as if they would stop instantly. In the same moment, the mainboard shuts down and wont turn back on until the battery has been disconnected and reconnected.  

I am currently using a nunchuk and my config's amp limit is set to 25.

Is this a short somewhere? A broken battery?

larsmm, 2018-05-27:
The i2c bus is very sensitive to emv distortions of the motor controller. They get stronger the faster you are. Keep cable short, use shielded cable, use ferrits, stabalize voltage in nunchuck, add i2c pullups. To many i2c errors leads to very high accelerations which triggers the protection board within the battery to shut everything down.

samgot1, 2018-06-06:
I have exactly the same issue. Always the same wheel is affected and it´s getting very hot.

closed 2018-11-10 by NiklasFauth