lucysrausch/hoverboard-firmware-hack #19: Explanation of the bldc.c-routine (Trapezoidal control 3-phase BLDC motors using hall sensors)

flo199213, 2018-05-25:
Hey guys,

I found an interesting article in my research of understanding what is done in the bldc.c-Routine to control the motor. If you want to unterstand the trapezoidal control of the 3-phase BLDC motors using hall sensors this might help you:

https://e2e.ti.com/blogs_/b/motordrivecontrol/archive/2013/11/08/generate-your-own-commutation-table-trapezoidal-control-3-phase-bldc-motors-using-hall-sensors

https://e2e.ti.com/blogs_/b/motordrivecontrol/archive/2013/12/20/generate-your-own-commutation-table-part-2

Especially look at part 2 where you find the explanation how the angle between the stator magnetic field and the rotor flux, could be set to 90° and the matching pwm-setting in the bldc.c-routine.  Attention should be paid to the fact that 50% pulse duty factor per phase means Aoff, Boff and Coff. That wasnt clear to me in the beginning.

Hope I could help you to understand faster than I did.

Florian

flo199213, 2018-05-25:
Yes, he explained it, but for me it was only clear in close contact with the article. I think the article is more near the code because it explains how to get from the hall states, to the hall to position table and to the PWM value setting. I assumed that there has to be some kind of u=pwm, v=-pwm, w=-pwm state for the six 60°, 120°, etc. edges. But the code will set in each position case one phase to 0 (u=0, v=pwm, w=-pwm) which represents the 30°, 90°, 120°, etc. edges. It wasn't clear to me,.... Maybe for some others who are not so deep in materia too.

closed 2018-08-09 by Jan--Henrik

oskarlove, 2018-08-11:
I'm still not getting it. How can 50% duty factor on a phase (pwm=0 -> duty value=1000) result in hi-off and lo-off (phase off)?