lucysrausch/hoverboard-firmware-hack #21: Erratic motors

rokrodic, 2018-05-31:
Hi. Does anyone have an idea why my motors start shaking like crazy when the speedL and speedR reach 390? I haven't tried negative values of their range -1000 to +1000.
I use the code below as "ADDITIONAL_CODE":

`
				if (button1 && !button2) {
					LEFT_TIM->BDTR &= ~TIM_BDTR_MOE;  //disable left
					RIGHT_TIM->BDTR |= TIM_BDTR_MOE;  //enable right
					pwml = 0; //pwml = -speedL;
					pwmr = speedR;
				}
				if (!button1 && !button2) {
					LEFT_TIM->BDTR |= TIM_BDTR_MOE;   //enable left
					RIGHT_TIM->BDTR |= TIM_BDTR_MOE;  //enable right
					pwml = -speedL;
					pwmr = speedR;

				}
				if (!button1 && button2) {
					LEFT_TIM->BDTR |= TIM_BDTR_MOE;   //enable left
					RIGHT_TIM->BDTR &= ~TIM_BDTR_MOE; //disable right
					pwml = -speedL;
					pwmr = 0; //pwmr = speedR;
				}
`

So, if I:
1. Disable right motor (button1 = 0 and button2 = 1) => left motor can go up to speedL = 1000 and right motor rests.
2. Disable left motor (button1 = 1 and button2 = 0) => right motor can go up to speedR = 1000, but left motor starts spinning too at speedR = speedL = 700 but pwml = 0; (not pwml = -speedL;) ????
3. This is for the first question in this post. In this case I am in the middle IF statement (ie button1 = 0 and button2 = 0). I see erratic shaking of the motors (as believe it's right motor, but hard to tell)...

I use:
#define FILTER              0.1  // lower value == softer filter. do not use values <0.01, you will get float precision issues.
#define SPEED_COEFFICIENT   1.0  // higher value == stronger. 0.0 to 1.0
#define STEER_COEFFICIENT   0.0  // higher value == stronger. if you do not want any steering, set it to 0.0; 0.0 to 1.0
...so no STEERING, CMD2 = SPEED = speedL = speedR in range 0-1000.

Thanks for any help.

larsmm, 2018-06-02:
Which input method do you use? ADC works best, nunchuck and ppm is very sensitive to emv distortions (see Troubleshooting)

rokrodic, 2018-06-04:
I use ADC:

      cmd2 = CLAMP(adc_buffer.l_rx2 - ADC2_MIN, 0, ADC2_MAX) / (ADC2_MAX / 1000.0f);  // ADC2 = SPEED
      //button1 = (uint8_t)(adc_buffer.l_tx2 < (2047-OFFSET));  // ADC1
      //button2 = (uint8_t)(adc_buffer.l_tx2 > (2047+OFFSET));  // ADC1
	  if (adc_buffer.l_tx2 < (2047-OFFSET))
		  button1 = 1;
	  else
		  button1 = 0;
	  if (adc_buffer.l_tx2 > (2047+OFFSET))
		  button2 = 1;
	  else
		  button2 = 0;
      timeout = 0;

Any ideas?

Parksy1234, 2018-06-17:
Hi,
Please see my comments at the bottom of the page, similar coding for mid point on ADC joystick and motor cutout for freewheeling using enable = 0 when joystick is near the centre , the buttons where causing me problems also.
https://github.com/NiklasFauth/hoverboard-firmware-hack/commit/2d2a9c4d11b850d5d6eda85854b2c2175170ffe8

rokrodic, 2018-06-18:
Thank you, will close the issue now. I have written my own debounce routines and do not use the ADC to falsify the buttons. I have pulled a merge request, so everybody can use it...

closed 2018-06-18 by rokrodic