lucysrausch/hoverboard-firmware-hack #25: Arduino control and power on

eduards-kazakovs, 2018-06-11:
I tried your new firmware with UART control.
Probably I'm doing something wrongly, but I can't get it working.
Would be great, if you could give some example on how to control it from arduino or similar, like very simple arduino script.
When I try to send something, behavior is very random - wheels spining in random directions at random speeds though each time command is sent, something changes.

Another thing is that with latest firmware power button has stopped working.
Only way how I can power it on is by connecting with openocd.

Thank you very much for the work you have put into this!

NiklasFauth, 2018-06-11:
Hey,

unfortunately I can't test the firmware, only reason I was publishing the UART control without testing it was because so many people were asking for it. Try using something like that:

```
void loop() {
    Serial.write((int16_t) steer); 
    Serial.write((int16_t) speed);
    delay(10);
}
```

For the power button, this can nit be caused by the firmware. Maybe your mainboard is broken, do you have another one to test?

Best

eduards-kazakovs, 2018-06-12:
You were right, apparently power button got broken somewhere in between. That's fixed for now.
I still haven't managed to control with arduino, seems that I'm getting very random behavior. Thing is that board reacts, but wheels just spinning at random speeds and random directions.

What are the speed and steer values board expects? What range and how the steer works.
Should there be some seperator between values and newline or return after the command?
Like what could I send from the Serial monitor to test that it's working properly?

NiklasFauth, 2018-06-12:
Hey,

the command is binary, so you can't just send values from the Serial Monitor.
You don't need any newline, packets are identified by timing.

As I said, I didn't test the firmware myself, I just uploaded it because there were like three people asking for it asap and I had it done for some time anyway, so I pushed it.
I can test it next weekend, until than all you can do is check that you use the correct baud rate etc.
Also you can try to disable the motors (set enable = 0 in the while(1)) and print the received command values to the other USART3 interface (like setScopeChannel(0, (int) command.steer)) so check if the values are received correctly. Maybe its also a problem of bitorder, but I don't have time to check that.

So, please wait 4 more days, or try to fix it on your own.

eduards-kazakovs, 2018-06-12:
Thank you for now, I'll wait for your test.
Managed to burn two Mosfets while testing.. :)

eduards-kazakovs, 2018-06-12:
I had the behavior where motors where going randomly at varios speeds. I think it burned when it suddenly switched from full speed in one direction to the oposite direction.
Maybe would be good to implement some kind of safety mechanism. Just thinking that maybe would be safer to have control over strings with delimiter instead of binary. Of course now it's shorter, but than it happens easily that even some noise can change the value.
I'm not expert on this, just noticed, that even touching the wire changes the speed.

eduards-kazakovs, 2018-06-13:
I got the whole thing working.
This is the arduino code what works for me:

`void loop(void){
  Serial.write((uint8_t *) &steer, sizeof(steer));
  Serial.write((uint8_t *) &speed, sizeof(speed)); //This splits value into the two bytes, otherwise only one byte is sent.
  delay(50); 
}`

At least for me "Serial.write((int16_t) steer)" was sending only one byte, 8 least significant bits.

eduards-kazakovs, 2018-06-13:
How it would be, if command would be sent in binary, but there would be additional checksum byte or two, for example sum of both values?
Just thinking that it could prevent that any random binary values trigger some action.

NiklasFauth, 2018-06-13:
Thanks for finding a working solution. I'm not into Arduino, for some reason I was expecting Serial.write to send all bytes... Well.

ASCII protocols are bullshit, it's not worth the overhead and also takes up more CPU.
The better way is to use CRC, I will implement that with an Arduino example code next weekend.

closed 2018-06-13 by NiklasFauth