lucysrausch/hoverboard-firmware-hack #27: Noise and vibration problem

HITMAnsOFT, 2018-06-17:
Hi! Has anyone seen the issue with loud noise and vibration from both motors, varying with RPM? The original firmware doesn't have this issue.
The motors start up very smooth, then as they pick up revs (from speed or steer value 300 or so) the whine gets louder and louder. It's not the PWM whine, it has much lower frequency, and if I touch the frame I can feel it vibrating. Then as it gets to full speed, the whine disappears.

I'm controlling it via UART.

HITMAnsOFT, 2018-06-17:
I've tried switching the motor wires to the alternative layout, that made it run much worse, so this is not the problem

threshold0420, 2018-06-17:
Hi, how you controlled it via UART?

HITMAnsOFT, 2018-06-17:
In Inc/config.h:
#define CONTROL_SERIAL_USART2
#define CONTROL_BAUD       115200

I'm controlling it from a STM32F103 bluepill board. Just sending out 4 bytes of steer and speed data. The vibration is proportional to the motor RPM, much like what a bad gearbox would make.

eduards-kazakovs, 2018-06-20:
I'm getting same kind of behavior. Controlling over UART as well.

rene-dev, 2018-06-21:
please check if you get the same problem with analog control.

rene-dev, 2018-06-21:
do you have a second board to test? could be one or more broken mosfets.

HITMAnsOFT, 2018-06-21:
No, just this one. Both motors are behaving the same. I have left the current limit at 15A. I'm trying to get used to how it runs for now, so that I can focus on developing my robot platform further. When turning slowly it's pretty smooth, no jerking, just some rumble is heard and felt. I tried holding the wheel back by hand on very low speed and found no interruptions in torque, so I think FETs are OK.

HITMAnsOFT, 2018-06-22:
I think my issue is pretty well described and analysed in #7, so I can close this.

closed 2018-06-22 by HITMAnsOFT

oskarlove, 2018-08-10:
I have the same issue. But I can't see an explanation in #7.
Some people says that there is no noise and some people says there is.
I understand that this firmware does not do FOC. But if that is the reason for the noise and vibrations, why isn´t everybody experiencing that?

HITMAnsOFT, 2018-08-10:
I can guess that it depends on the quality of the original firmware that people compare this with. If someone uses an older board with a simple firmware which works the same way, they will not report any difference. Newer firmware however might be more sophisticated, therefore smoother.

oskarlove, 2018-08-10:
There is no way that any version of the original firmware has this noise. Under "heavy" load (meaning driving a human person on a flat floor with pwm 50%) there is a very loud "cogging" noise and vibrations. I have bought maybe 10 hoverboards over the last 4 years (for the parts) and I have seen 3 types of control boards. All of them was silent even under extreme load.
Maybe the people that says that there is no difference in noise between the original firmware and this firmware didn't test it with a load.