lucysrausch/hoverboard-firmware-hack #28: Sensor board firmware

kloppertje, 2018-06-20:
Hi,

Currently, I'm working on custom firmware for the (two) sensor boards. In its current status, it has been tested working on one of the sensor boards (reading the MPU6050 and sending data over serial). The format is not the "original" format (4 bytes at 26300 baud), as its purpose is to communicate with the custom mainboard FW.

My purpose for this sensor board FW would be to run a PID controller on the hoverboard, probably this controller ends up in the sensor board in a first implementation, sending speed (=pwm) commands to the mainboard with custom FW.

Would there be any interest from others for this code? If so, please let me know. 

Some of the future possibilities:
- Read the optical sensors
- Control hoverboard LEDs
- Make the hoverboard self balancing

LeoDJ, 2018-06-20:
Funnily enough, Niklas started working on this too: https://github.com/NiklasFauth/hoverboard-sideboard-hack

NiklasFauth, 2018-06-20:
Hey,

please note that the firmware Leo refereed to is for sideboards that use a GD32F130 CPU + ICM-20618 IMU. This is a very unhappy combination, however this is what almost all 20 sideboards that are currently laying around from previous hoverboard hacks are using. 

However I would be very happy to see a firmware for the not-so-common but STM32f103 based sideboards. My current protocol uses: 

```
typedef union {
  float f;
  char  s[4];
} floatToByte;

typedef struct{
  uint8_t header;
   floatToByte roll;
   floatToByte pitch;
   uint8_t status;
   uint8_t endbyte;
   //uint32_t crc;
} Balancecommand;
```
so 11 bytes sent over 38400 8N1 UART with 200Hz.

If you want to discuss further about your software, you can also join our Telegram group:
https://t.me/joinchat/BHWO_RKu2LT5ZxEkvUB8uw

Best,
Niklas

kloppertje, 2018-06-20:
The board indeed uses an STM32F103C8T6 (which is very commonly used on red/blue/black pill development boards). 

Your proposal for the protocol, is it already in use somewhere? I believe I haven't seen this structure earlier in your mainboard code. Initially, I was planning on using the basic UART speed communication currently implemented in your mainboard FW. That way, the sensorboard can do all the calculations (PID control, receiving commands from e.g. a remote control), while the main board manages the motor control.

The code I've developed is quite basic (which is good as a starting point), making use of the HAL libraries. It has been generated with STM32cubeMX, which allows for easy customization of interfaces/inputs/outputs etc. 

I'll make the repo public in some time, after I've developed the code a bit more.

HITMAnsOFT, 2018-06-21:
Just as a note on possible MCU/IMU combinations, I have a GD32F130 with an MPU6050 :) @NiklasFauth , do you know the STM equivalent for the GD32F130 or are you using GD-specific HAL package?

rene-dev, 2018-06-21:
there is no hal lib or st equivalent for the gd130. you can download the refman, datasheet, cmsis lib, and stdpheriph lib from the gigadevice website.

HITMAnsOFT, 2018-06-21:
That's what I suspected. Thank you for info!

kloppertje, 2018-06-24:
I've made the sensorBoard code public, see https://gitlab.com/kloppertje/hoverboard-firmware-hack/tree/develHW/sensorBoard

Please note that the code is currently in a preliminary state. The sensor board sends steer/speed data to the mainboard via uart2. The angle obtained from MPU6050 is used as speed command. Currently, I'm using the hoverboard lying on it's back (for testing), just so you know.

RomSand, 2018-09-17:
I want also to start experementing with sideboards but can't download original FW, Does somone has copy of original firmware ? Thanks.

closed 2018-11-10 by NiklasFauth