lucysrausch/hoverboard-firmware-hack #30: Motor issues - Erratic / Uncontrollable Wheel / Power Lock Out

flamnitka, 2018-06-22:
(firstly - love the project, thanks to all contributors).
I have some unexpected motor issues, all relating to motor operation.

**Environment:** Firmware flashed via Ubuntu using described flash method. My hoverboard appears to be the same PCB as the PCB shown on the main project page [here](https://github.com/NiklasFauth/hoverboard-firmware-hack) - markings on my PCB are `679394 ROHS 20150722`

I have flashed with a variety of combinations, toggling `define DEBUG_SERIAL_USART3` and `define DEBUG_SERIAL_USART2` and have tired multiple combinations of uncommenting various CONTROL and DRIVING BEHAVIOR settings as per the [config.h](https://github.com/NiklasFauth/hoverboard-firmware-hack/blob/master/Inc/config.h) file.

*Issues:*

Note: The below issues have been observed while the Nunchuck is directly connected to the Hoverbaord (via jumper leads), and also when the Nunchuck is connected to the [Breakout PCB](https://github.com/Jan--Henrik/hoverboard-breakout). I understand there is EM interference from the motors, however this seems like a firmware issue, not a control/interference issue.

1. **Nunchuck - uncontrollable LEFT WHEEL:** While the hoverboard is placed upside down (ie wheels are freely spinning in the air), when powered on the Nunchuck will correctly control the motors for a few seconds, then the RIGHT WHEEL will stop and the LEFT WHEEL will go into full throttle mode and not stop. Both wheels are now totally unresponsive to the Nunchuck control and the only way to stop the wheel is to turn the power off by pressing the power button.

2. **Nunchuck - Unit lunges erratically, then stops**: While the hoverboard is placed correctly on the ground (ie wheels are in contact with the ground and there is a load/person weight on the Hoverboard), when powered on the Nunchuck will correctly control the motors for LESS THEN 1 SECOND and then the hoverboard will either lunge forward or backwards for another second and then become unresponsive, the motors will shut down and require a restart to become functional again.

3. **Unit unable to turn off:** - This has only occurred once but was quite a serious issue. The unit was turned on, Nunchuck connected via cable to the Hoverboard, the wheels were on the ground and a person was on the hoverboard. The issue as per no. 2 above occurred (power on, control via Nunchuck for few seconds, unit lunged forward) however in this case the unit kept supplying power to one wheel, and the power button was completely unresponsive. The wheel started getting hotter and hotter, it couldn't freely spin (it's like it was being supplied power from all directions). The only solution was to unscrew the unit and disconnect the butter connection.

Thanks for you help in find a solution - and again thanks for the project!

kloppertje, 2018-06-28:
Could you maybe gather some data? You can send a few bytes to uart2 or uart3, which you can then read with a serial monitur (using a usb to serial converter). By default speed1 and speed2 are logged. If you check those, you can see if the issues arise from commication issues. If not, you can debug further by plotting different values.

flamnitka, 2018-06-29:
Thank you kloppertje,

Below is the output I recorded using an Arduino via it's serial monitor.
I have enabled `#define CONTROL_NUNCHUCK`. An interesting issue is that I'm only able to view the serial output for a few seconds before I receive ⸮⸮⸮⸮⸮⸮⸮⸮⸮~⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮ and the monitor freezes (tried using 2 different Arduino).
I have noticed that during the 'make' I get some warnings and errors - I've assumed they're OK and just gcc spitting out warnings such as 'warning multi line comment' and 'implicit declaration of function' - do you get these too?

> 0;0;259;259
> 0;0;259;259
> 0;0;259;259
> 0;0;259;259
> 0;0;259;259
> 0;0;495;495
> 0;0;495;495
> 0;0;495;495
> 0;0;495;495
> 0;0;495;495
> 0{⸮⸮⸮⸮⸮⸮⸮⸮⸮~⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮

flamnitka, 2018-11-28:
Hey @m-farahmand
Interesting we both have the same issue, maybe it's unique to our hardware?
I'm attaching a picture of my board - it has the markings "679394 ROHS 2015072"
Anyone else have this issue?
[image]

p-h-a-i-l, 2018-11-29:
What kind of input method are you using? From my experinece, ADC is the most reliable Method. Nunchuck is unstable and UART very dangerous (no checksum, easily out of sync but no way for the controller to know. Gibberish will be interpreted as speed and steer inputs). Haven't tried PPM, but to check if the motor controls are working, I'd try with an analog voltage from 0-3.3V for the ADCs.

Regardings switching off: When the wheels are still turning, the off button just stops the wheels in my units. No shutdown melody, no more reactions. Battery disconnect is necessary to get it up again.

m-farahmand, 2018-12-01:
@flamnitka, My board is exactly as the same as yours. I changed the firmware to [this](https://github.com/larsmm/hoverboard-firmware-hack-bbcar) and it worked. Thanks to @larsmm.

vmaliaev, 2019-04-14:
Hey guys! Is anyone of you please have a chance to measure voltage on leg 15 (AIN_V_BUTT) of the MCU when the power button is pressed. In my case it is 1.35V. It seems that capacitor C5 (according to connectors.sch schema) is damaged and Voltage is not high enough to set the logical high.
I'd appreciate your help.

vmaliaev, 2019-05-01:
It seems that everything is ok, this is just a circuit scheme that has been made in such a weird way.

ashwin-ar, 2020-02-26:
hey, @p-h-a-i-l,
1)I have a similar kindof problem .I couldn't turn off the board again by using the power switch.
2)and also I am using a UART communication between the board and arduino. The problem is that if once the value is given in the arduino the board runs in the same speed continuously and also if the speed is changed it doesn't responds to the change.I tried making it zero and update the value again for every 25 loops but it too didn't work out.

p-h-a-i-l, 2020-02-27:
@ashwin-ar Sorry, haven't followed the UART communication implementation in this repo