lucysrausch/hoverboard-firmware-hack #31: Motors shaking when controlling from UART

kloppertje, 2018-06-27:
I'm currently controlling the motors via uart2/3 (adapted code to work with either of those). In general, this works properly: speed commands are read correctly. However, sometimes, after startup the motors start shaking heavily. I don't believe the sending part goes wrong. 

My current hypothesis is as follows. The shaking either happens continuously, or not at all. So, it is not happening because of inaccurate data (missed bits/bytes). Four bytes are sent as command. However, if the byte that is read is not the first one that was sent, the bytes are read shifted. E.g. normally speed least significant byte is read correctly. However, now speed lsb is read as speed msb; lsb is noisy, and so is motor speed. 

Could this be correct?

rene-dev, 2018-06-27:
it is not a good idea to use uart without something like crc. Sounds like a framing problem.

kloppertje, 2018-06-27:
@NiklasFauth I'm currently implementing idle detection for uart communication in the sensor board. Diy you look into this for the main board? It is not supported by the HAL libraries by default, but would prove useful here I think.

closed 2018-11-10 by NiklasFauth