lucysrausch/hoverboard-firmware-hack #32: Mainboard blew up

walterinho, 2018-07-04:
Hi there,

I did flash your code to my board successfully, but it blew up once I connected it.

[image]

I don't know what was the problem, do you think I could replace those components?

rene-dev, 2018-07-04:
you probably flashed unsupported hardware. post a pic of the rest of the board.

walterinho, 2018-07-04:
This is the mainboard.

[image]

I do have another board (see below), but the st-link tool says there's no target connected:

[image]

Is there any other way in which you could please help me with some technical advice? it would only be a few minutes

walterinho, 2018-07-10:
@rene-dev , How can I know if the hardware is supported?

I have this new mainboard, and I don't want to screw it up again.

[image]

LeoDJ, 2018-07-12:
The best way to test it is to use a different power supply than the battery, that is current limited. Something like a lab bench power supply or if you feel adventurous and skilled enough the supplied hoverboard charger. But I highly recommend a lab bench power supply.
If the firmware on the mainboard tries to pull too much current, it will be limited by the supply instead of blowing components up with the high current from the battery.

HITMAnsOFT, 2018-07-14:
Is that the buzzer control transistor? It shouldn't be fatal to the operation of the board. You can still try to run it with a current-limited power supply.

johnsokol, 2018-07-19:
I have the tools to fix boards at the hackerdojo.

walterinho, 2018-07-19:
Thank you very much for your help guys, please bear with me, I'm a noob at some topics in Electronics.

@LeoDJ, thanks! I followed your instructions with the new board and it works perfectly, I tested it with a maximum of 1.5A. Now my problem is that I need to use the hoverboard's battery, do you think that after doing these test it is safe to use it to the original battery with this board? or what options could I have?

@HITMAnsOFT yes, it is the buzzer control transistor and I thought the same but it doesn't work at all, I tried to test what is going on by connecting the oscilloscope to each motor phase and GND (I don't know if this is correct haha) and I realized that one of the phases of each motor doesn't change it's width (duty cycle) when I send a different speed and steer command. Is there a better way of checking what is wrong?

@johnsokol thanks for your help, the problem is that I live in London, =(, but I really appreciate that you are interested in helping me, maybe you have any advice about how to check what is wrong with the board.

btsimonh, 2018-08-29:
@walterinho I have the same YST board, and all is working well.

closed 2018-11-10 by NiklasFauth