lucysrausch/hoverboard-firmware-hack #33: Sinusoidal control of PMSM

r2aiv, 2018-07-06:
Hi Niklas! What do you think about rework your project to sinusoidal control of PMSM? This may be enhance motor torque and speed, and motor will works better! I want to rework your project to this, but have some questions about scooter PCB and your code. Can you help me? How i can contact you over email?

LeoDJ, 2018-07-12:
You can ask the questions here as well, or even better in the [Telegram Group](https://t.me/joinchat/BHWO_RKu2LT5ZxEkvUB8uw). That way, more people that knows about the details can help you as well.

oskarlove, 2018-08-10:
Hi.
Any progress in this?

I think sinusoidal commutation is necessary to get the same smooth drive as the original firmware. I get a "cogging" noise when driving with heavy load. And I get loud vibrations on some speeds.

btsimonh, 2018-08-25:
yep; just competed implementation of original sensor drive; the hoverboard now works again as a 'hoverboard'; but it does 'grind' a bit.

AerDronix, 2019-05-21:
@btsimonh are you still active on this topic? I saw in your repository that you did something on sinusoidal control. Is that functional?

btsimonh, 2019-05-21:
no, but looking for someone with a brain :).
Repos at https://github.com/bipropellant
I really want to have very slow control.  Theory says that PID control should be able to achieve 6mm, but it's a lot more sloppy than that at the moment.  Repos updated today....

AerDronix, 2019-05-22:
@btsimonh Ok. Because I am working on improving the motor control right now. It looks promising until now. I decided to do it, because I see the commutation method from Niklas vibrates a lot. I am a Control Engineer and that is not acceptable in my opinion. I hope in like about 2 weeks to have something that I can publish.

By the way, I also flashed your modification. I wanted to re-enable the hoverboard mode. However, I did not manage to make the motherboard read the child boards. Maybe I have a different board than you do... I admit, I did not spend time on debugging.
At the moment I am using the repository from Niklas. After I finish with motor control I will have another look on your implementation.
Do you know what is diferent in the Repo https://github.com/bipropellant compared to yours?

btsimonh, 2019-05-22:
we moved my to bipropellant as the other two were doing more dev than me :) .... and split the repo to have a sub-repo for the protocol aspects (because we anticipate using it in the controllers as well...).
Note this [issue](https://github.com/bipropellant/hoverboard-firmware/issues/10) - @lalalandrus is also trying to get the sensor boards to talk to an STM based board.  Seems he now has lights, but the sensor data interpretation or usart code is not yet functional on STM based boards.  Checked mine last night with latest master branches, and it's fine, but I have a YST GD32 based board. (build with platformio and specifically build the hoverboard env).
The BLDC does 'grind' a bit, and I also get some 'clicks' in use, worrying but they do not adversely affect operation as a hoverboard.
Looking forward to your thoughts and implementation!

EmanuelFeru, 2019-05-22:
Quick response :) thanks. Yes, I have and STM board (STM32F103RCT6 on the main board) and GD32 on the child boards I think (I don't have the board with me right now). Would be nice if @lalalandrus could fix the communication issue on STM32 boards.
Then we could combine the new control with your implementation. I think it should be very close to the orginal firmware in terms of noise and performance but with much more flexibility.

lalalandrus, 2019-05-22:
fine! just so happens i fixed it an hour ago though it is a sweeping change, ill upload it to my fork only and @btsimonh can have a look to see if he wants a pull, it fixes having to send the LED's twice too

EmanuelFeru, 2019-05-22:
@lalalandrus Great! I would like to try it. Because do you know what I did to save the orginal firmware? 
I ordered few STM32 chips on ebay https://www.ebay.com/itm/1PCS-5PCS-STM32F103RCT6-LQFP-64-STM32F103RCT6TR-STM32F103RCT-MCU-ARM-ST-IC/183647636767?ssPageName=STRK%3AMEBIDX%3AIT&var=691112271463&_trksid=p2060353.m2749.l2649 
Then I desoldered the original STM32 and resolder the empty one... actually it was a good idea because I can go revert the board whenever I want. Maybe I should make a video or something because it came one quite nice.

And yes, field weakening will be definitely there :) as I want to reach high speeds as well (if needed) because I might use these motors for other applications (as a motor to spin some other stuff).
Let's keep in touch, I will come back to you with updates.

lalalandrus, 2019-05-22:
btw i too am controls engineer

EmanuelFeru, 2019-05-22:
Cool, then we can exchange some ideas if I get stuck somewhere

btsimonh, 2019-05-22:
haha! 'desoldered the original STM32' - I took a software approach and tried to apply the hacks on the STM debug interface to extract it, but seems the GD32/STM32F103 are not susceptable :(.

EmanuelFeru, 2019-05-22:
yeah, I went the hard way... haha, a bit of work and risk but luckly it worked. I read somewhere that code can be extracted, that whould be a more elegant solution. Is a pitty that it cannot be extracted :(

lalalandrus, 2019-05-22:
ok check my repo now

p-h-a-i-l, 2019-05-22:
@EmanuelFeru Did you have a look at https://github.com/BobbyIndustries/hoverboard-firmware-hack for BLDC? When I made an overview of different branches some time ago, this was the one with most modifications in the BLDC part. Didn't have the time to look deeper, but you can see keywords like sonsorless in the code.

EmanuelFeru, 2019-05-22:
► Thanks. I will try it these days. My baud rate is a strange one 32100 bps.
► I have seen that repo. Maybe they use better protection algorithms to protect the system. So, I think is worth having a look. However, I see they still use the commutation method (in bldc.c file), so I expect more or less similar griding noise. I will check it out.

lalalandrus, 2019-05-22:
that is really odd i want to ask the question of what is your pulse width

EmanuelFeru, 2019-05-22:
hmm... if I rember correctly is was something around 3.1 us which leads to ~32100 bps

lalalandrus, 2019-05-22:
3.1 doesnt seem right 23000 is 38us

EmanuelFeru, 2019-05-22:
31 us sorry. So 1/(31us) =~ 32100 bps somwhere there

btsimonh, 2019-05-22:
@lalalandrus 
> ok check my repo now, grab the new config.h again to get the proper baud rate too

looking good.  Surprised it ever worked with the bug in USART_sensor_addTXshort!  pop in a PR and I'll test on mine this evening....

lalalandrus, 2019-05-22:
Would be cool if we had vector control which would get field weakening and precise rpm control at the same time. We should also move the implementation discussion over to the other fork so we not spamming unrelated forks and reviving year old zombie threads 🤣

EmanuelFeru, 2019-05-23:
@lalalandrus
>>Vector control like FOC, indeed would be nice, however I think we will reach the STM32F1 Series computational limitations, since we need to run the algorithm at 100 kHz and plus some additional code for balancing controller or other calculations. STM32F4 series are much more capable but they are not in the hoverboard.
>> About moving the discussion to other fork... Do you know how to do it? I am still new in github. I am more used to svn.

btsimonh, 2019-05-23:
@EmanuelFeru - just continue it here:
https://github.com/bipropellant/hoverboard-firmware/issues/10
or start a new issue about BLDC improvements by going to that repo and clicking 'new issue' :).

EmanuelFeru, 2019-05-23:
As OK. Thanksn

EmanuelFeru, 2019-06-10:
Sinusoidal control of the BLDC is implemented here:
https://github.com/EmanuelFeru/hoverboard-firmware-hack
Enjoy!

closed 2020-03-20 by Jan--Henrik