lucysrausch/hoverboard-firmware-hack #34: LCD code not working.

o0Mouse0o, 2018-07-25:
Hi, 
I have tried to enable the LCD code by de-commenting the IFDEF flag in config.h 
#define DEBUG_I2C_LCD
After this the code does not compile. 
I am at a bit of a loss as there are a lot of HAL calls involved with the LCD of which I am unfamiliar. 

The compile error I get is this.
==================================
Src/main.c:143:5: error: 'lcd' undeclared (first use in this function)
     lcd.pcf8574.PCF_I2C_ADDRESS = 0x27;
     ^~~
Src/main.c:143:5: note: each undeclared identifier is reported only once for each function it appears in
Src/main.c:146:29: error: 'NUMBER_OF_LINES_2' undeclared (first use in this function)
       lcd.NUMBER_OF_LINES = NUMBER_OF_LINES_2;
                             ^~~~~~~~~~~~~~~~~
Src/main.c:147:18: error: 'TYPE0' undeclared (first use in this function)
       lcd.type = TYPE0;
                  ^~~~~
Src/main.c:149:10: warning: implicit declaration of function 'LCD_Init'; did you mean 'HAL_Init'? [-Wimplicit-function-declaration]
       if(LCD_Init(&lcd)!=LCD_OK){
          ^~~~~~~~
          HAL_Init
Src/main.c:149:26: error: 'LCD_OK' undeclared (first use in this function); did you mean 'HAL_OK'?
       if(LCD_Init(&lcd)!=LCD_OK){

==================================

Parksy1234, 2018-07-26:
4 files missing defined in makefile and main.c
you can find them in the github hoverboard_hack repository

hd44780.c
hd44780.h
pcf8574.c
pcf8574.h

i seem to remember using an i2c file as well 
look back to where they said it was working
and dont forget your pull ups and resistor bridge for the 3,3 to 5v conversion.

main,c
#include "stm32f1xx_hal.h"
#include "defines.h"
#include "setup.h"
#include "config.h"
//#include "hd44780.h" //file was missing for lcd

void SystemClock_Config(void);

extern TIM_HandleTypeDef htim_left;
extern TIM_HandleTypeDef htim_right;
extern ADC_HandleTypeDef hadc1;
extern ADC_HandleTypeDef hadc2;
extern volatile adc_buf_t adc_buffer;
//LCD_PCF8574_HandleTypeDef lcd; //file was missing for lcd
extern I2C_HandleTypeDef hi2c2;
extern UART_HandleTypeDef huart2;

and in makefile
drop the c file paths in here
# C sources
C_SOURCES =  \
Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_flash.c \
Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_pwr.c \
Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_rcc.c \
Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_tim.c \
Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_tim_ex.c \
Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_gpio_ex.c \
Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_adc_ex.c \
Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_cortex.c \
Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_flash_ex.c \
Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_gpio.c \
Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_rcc_ex.c \
Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal.c \
Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_adc.c \
Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_uart.c \
Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_i2c.c \
Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_dma.c \
Src/system_stm32f1xx.c \
Src/setup.c \
Src/control.c \
Src/main.c \
Src/bldc.c \
Src/comms.c \
Src/stm32f1xx_it.c \

have fun, the best i got was its initialising version 2.....

zsocietyGithub, 2018-09-21:
> 4 files missing defined in makefile and main.c
you can find them in the github hoverboard_hack repository
hd44780.c
hd44780.h
pcf8574.c
pcf8574.h

i can't find this libb ....can you give specific link the repro?
i use VScode IDE using platform.io can i use library in platform io library list?

thanks