lucysrausch/hoverboard-firmware-hack #36: Board halting if speed is close to 0

duncan999007, 2018-07-28:
I have everything else working, communication, motors, all the good stuff. I have one problem though, in that if I go from -100 to 100 speed, the motors will stop and the board will become unresponsive until restarted. If I'm running it on PPM, 1400 will reverse, 1600 will forward, but going from one to the other doesn't happen. Setting a value anywhere between 1450 and 1550, or close to -50 and 50 speed, including 0, will cause the motors to stop and the board will become unresponsive. I've tried changing current limit, batteries, bench power, disabling and enabling debug settings, I'm out of ideas. The debug monitor just stops as soon as the motors do.

Parksy1234, 2018-07-29:
In main.c this can cause problems because of timeout, i comment out the if and the keys{} but keep the commands or there's no output
    // ####### SET OUTPUTS #######
   if ((speedL < lastSpeedL + 50 && speedL > lastSpeedL - 50) && (speedR < lastSpeedR + 50 && speedR > lastSpeedR - 50) && timeout < TIMEOUT) {

 if the PPM is flakey the timeout is changed in control.c

Have you used long cables, 3.3v, resistor dividers, any filter capacitors and pull ups. ?

duncan999007, 2018-07-29:
Short cables directly to an Arduino running a PPM generator on an interrupt timer. Also tried using an unbranded RC receiver, same thing. I'll try removing that if statement in a bit and see what that does

blubbi321, 2018-09-14:
@duncan999007 did you get it to work in the meantime?

duncan999007, 2018-09-22:
I never did. Put it on hold until someone smarter than me had the same issue

snowkingcro, 2019-01-28:
Hi @duncan999007 
As it seems that only you have tested the PPM pin here, wondering if you could point me right...
I'm planing to hook up a 2,4Ghz RC Receiver direct to the board, do you think it could work?
I didn't try it yet, as i'm trying to compile the changed config.h 
If it happens that you have already compiled the hex, would you share it somehow?

duncan999007, 2019-01-28:
I won't share it, just out of licensing liability, but it should work with
the RC receiver if you have the code set up right

snowkingcro, 2019-01-28:
That is Ok. 
But, can you point me out if i have to configure something else, other than changing 
control in config.h?
Thank you in advance.

// ###### CONTROL VIA RC REMOTE ######
// left sensor board cable. Channel 1: steering, Channel 2: speed.
#define CONTROL_PPM                 // use PPM-Sum as input. disable DEBUG_SERIAL_USART2!
#define PPM_NUM_CHANNELS 6          // total number of PPM channels to receive, even if they are not used.

// ###### CONTROL VIA TWO POTENTIOMETERS ######
//// ADC-calibration to cover the full poti-range: connect potis to left sensor board cable (0 to 3.3V) (do NOT use the red 15V wire in the cable!). see <How to calibrate>. turn the potis to minimum position, write value 1 to ADC1_MIN and value 2 to ADC2_MIN. turn to maximum position and repeat it for ADC?_MAX. make, flash and test it.
#define CONTROL_ADC                 // use ADC as input. disable DEBUG_SERIAL_USART2!
#define ADC1_MIN 0                // min ADC1-value while poti at minimum-position (0 - 4095)
#define ADC1_MAX 4095               // max ADC1-value while poti at maximum-position (0 - 4095)
#define ADC2_MIN 0                // min ADC2-value while poti at minimum-position (0 - 4095)
#define ADC2_MAX 4095               // max ADC2-value while poti at maximum-position (0 - 4095)

ashwin-ar, 2020-02-26:
hey, @duncan999007 I have a similar kind of bug,I am using a UART communication between the board and arduino. The problem is that if once the value is given in the arduino the board runs in the same speed continuously and also if the speed is changed it doesn't responds to the change.I tried making it zero and update the value again for every 25 loops but it too didn't work.