lucysrausch/hoverboard-firmware-hack #37: Did anyone look at the hoverboards with two identical separate controllers?

btsimonh, 2018-07-29:
I have a hoverboard with two separate controllers (image below).

Did you consider this design of controller at any point?
They seem to produce a 200kbaud output (not fully analysed yet, so not sure if they can be controlled).
The firmware is protected; can't be extracted.  I am on the lookout for a suitable binary file if anyone has one - so far nothing has come up in google.
I have done a basic outline of the MCU connections; if anyone is interested, I could write this up. 

[image]

flo199213, 2018-07-29:
I had a look at the two Mainboard device:
https://github.com/flo199213/Hoverboard-Firmware-Hack-Gen2
Including schematics, custom firmware, etc. Still not finished. I have a newer finished version on my computer. Will upload it soon.

btsimonh, 2018-07-29:
thanks florian, the MCU pinout does not exactly match mine, but I torched one of my boards, so have ordered a new pair from ebay; maybe they will match yours :).
Will, check out your repo in detail :).

Jan--Henrik, 2018-08-09:
We have no firmware blobs for these boards, nor do we write firmware for them. Thanks @flo199213 for pointing out that there is one for these boards.

As far as I know these boards are "generation 1" while the boards with two drivers per board (that we use) are "generation 2".

btsimonh, 2018-08-10:
Hi Jan, 
funny; my ebay boards arrived, and the seller sent me the wrong ones, so I'll be following this repo after all.  However; the board (black, YST) that arrived is a little 'updated' with sites for what I assume is bluetooth audio on the sensor boards, and the chinese GD32F103xx CPU rather than STM, plus a couple of extra connectors ('AD'? and a power outlet).  I'll fork and add board images and information as time allows.  For my (robot) application, I'll be looking to fine positional control rather than speed control... 

Is there any original firmware for the combined main board or the original sensor to be found?

drbytes, 2018-08-21:
I have one of these boards and was able to flash it after unlocking it. Now I need to figure out how to get it to talk to a Pi. Any pointers?

p-h-a-i-l, 2018-08-22:
@drbytes, I guess that this question is not related to the old hardware revison discussed in this Issue? Please open a  new Issue to keep topics clean.
As a pointer, have a look at the `// ###### CONTROL VIA UART (serial) ######` area in  config.h, I think this is a good starting point.

tequilahut, 2019-04-16:
does anyone have any experience with this board? is this as flat as the others? it's obviously a gd32 chip. i think the pinout to flatten the chip is the one that says:PROHIBIT. the third pad from the left is also GND which actually speaks for it. 
[image]
[image]

tequilahut, 2019-04-17:
> 
> 
> I have one of these boards and was able to flash it after unlocking it. Now I need to figure out how to get it to talk to a Pi. Any pointers?

do you know the pinout of the "prohibit" port? is it 3,3v swc gnd swdio? can you share some informations? what kind of software did you use?

tequilahut, 2019-04-17:
@drbytes can you share some informations? that would be awesome!!

drbytes, 2019-04-17:
I have up on this. Plan was to buil a lawn mower but shortly after I had to replace the mower with a real automower. Project has been sitting in a box ever since :/

tequilahut, 2019-04-26:
@drbytes ok thanks. but you flashed those boards already? can you tell me the pinout of the flashins?

> I have up on this. Plan was to buil a lawn mower but shortly after I had to replace the mower with a real automower. Project has been sitting in a box ever since :/

[image]

is it like this? what programs did you use to unlock those boards?

tequilahut, 2019-04-27:
ok I got the pinout from this website: [http://www.stm32duino.com/viewtopic.php?t=3485&start=220](url)
... here my update.... 

[image]

I am using this ST-Link V2 adapter...: [https://tenbaht.github.io/sduino/hardware/flashtool/](url)

1 (left) - SWDIO
2 - SWDCLK
3 - GND
4 (right)- VDD

now I can connect via ST-link v2... but not I need to disable the read out protection.....

tequilahut, 2019-04-29:
> I have a hoverboard with two separate controllers (image below).
> 
> Did you consider this design of controller at any point?
> They seem to produce a 200kbaud output (not fully analysed yet, so not sure if they can be controlled).
> The firmware is protected; can't be extracted. I am on the lookout for a suitable binary file if anyone has one - so far nothing has come up in google.
> I have done a basic outline of the MCU connections; if anyone is interested, I could write this up.
> 
> MCU is GD32F103C8T6, sensor is marked Mb81 (?) but appears to match an MCU-6050 pinout.
> 
> [image]

@btsimonh could you please share your MCU pinout?

qwqw330, 2020-03-25:
Hello, I do have such a dual mainboard hoverboard too. Was any progress made on these? It has a gd32103c8t6. Looks like i have unlocked the flash, but didnt get much further. Has anyone tried to get a pinout yet?

[image]

andresiko, 2020-07-22:
> Hi Jan,
> funny; my ebay boards arrived, and the seller sent me the wrong ones, so I'll be following this repo after all. However; the board (black, YST) that arrived is a little 'updated' with sites [...]

Did youmanage to flash those boards?

tequilahut, 2020-08-10:
> Hello, I do have such a dual mainboard hoverboard too. Was any progress made on these? It has a gd32103c8t6. Looks like i have unlocked the flash, but didnt get much further. Has anyone tried to get a pinout yet?
> 
> [image]

Hi did you went any further? its so hard to start from scratch... I have not a single clue what to do....

tequilahut, 2020-08-10:
> Did youmanage to flash those boards?

I know that I flashed Them and kind of unlocked them... thats all that I've got so far.. at the moment I don't know how to move on from here.... idk the enviroment and how to start...

qwqw330, 2020-08-10:
No sorry. I didn't really have the time and patience to work on it.
I just got out my current work (with the original mainboard) a few hours ago. Maybe i'll take another look at the small boards.

The last thing I remember about my boards was that on one side 3 mosfets were dead and I didn't know how much else was affected, but on the other I could write the flash, power of the device, reboot it, read the flash and it was exactly the one I wrote onto it. So it did kinda work, but that's all i had.
The pinout can't be _that_ much different though, atleast for the timers ...

JohMah, 2021-09-01:
I have the dual boards that look very similar. Managed almost to more ore less reverse engineer it. Schematic in Eagle. In 
case someone is still interested.
[image]

phpkadir, 2021-09-02:
JohMah can you share the detailed schemantics of this cards. if i can solve schemantics i can write a firmware code.

JohMah, 2021-09-02:
This is the current status.
[Hoverboard JC2015.7.31.pdf](https://github.com/NiklasFauth/hoverboard-firmware-hack/files/7101731/Hoverboard.JC2015.7.31.pdf)

benhadad, 2022-02-22:
I have a similar model. Wanted to add to the thread and highlight mine had a remote control module plugged into the External-Interface area.  I have included a picture of it. 
[image]
[image]

Ability-dis, 2022-03-19:
> I have a similar model. Wanted to add to the thread and highlight mine had a remote control module plugged into the External-Interface area. I have included a picture of it. [image] [image]

Hi thanks for sharing .I also have similar board. I was also searching about the "External Interface" function. Can you provide some details about this .Did you purchased the module from market or is it come with the hoverboard?. Can you show the remote and what functions are include in the remote.

I want revers forward and break function.

larsmm, 2022-03-22:
Hi, this project is deprecated. Have a look at this project:
https://github.com/EFeru/hoverboard-firmware-hack-FOC

benhadad, 2025-04-17:
> Hi thanks for sharing .I also have similar board. I was also searching about the "External Interface" function. Can you provide some details about this .Did you purchased the module from market or [...]

It came with the hoverboard I was taking apart.