lucysrausch/hoverboard-firmware-hack #40: wrong direction of stlink pins

hexxter, 2018-08-03:
Hi,
i have changed SWDIO and SWCLK and i see smoke. After if have changed this and programmed my pcb always after push the power button it beebs always in 0,5sec pulses.

Have anyone an idear what part form my pcb is dead?

BR Hexxter

LeoDJ, 2018-08-03:
Reversing the clock and data lines of the programmer should not produce a hardware error. Are you sure you didn't reverse anything else?

hexxter, 2018-08-03:
hi,
i'm not sure but i saw smoke and pull out the power cord. but after changing this pins in the right direktion. i can flash the pcb. and ich can control the motor via UART. But when i enable the Nunchuck then the pcb beebs always after switch it on.

BR Hexxter

Jan--Henrik, 2018-08-09:
@hexxter did you disable UART while enabling I2C?

hexxter, 2018-08-09:
i commt this out //#define DEBUG_SERIAL_USART3

hexxter, 2018-08-09:
i have used the PCB with a 3v3 regulator but i think the pcb has an bug i see 10V on the nunchuck power pins. I have downloaded the new version from the PCB (the regulator was turned 90 °) and ordered a new pcb. i hobe ma nunchuck is not dead after this...

Jan--Henrik, 2018-08-09:
Then you had an old version of the PCB. Probably the mainboard and/or the nunchuck is broken. Use a lab power supply for testing, but do not brake, as it is not regenerative.

closed 2018-08-09 by Jan--Henrik