lucysrausch/hoverboard-firmware-hack #43: building in platform.io

btsimonh, 2018-08-11:
Hi All, 
I've just spent a couple of hours getting this to build in Platform.io.  It now produces a .bin and .elf file

The modifications are here:
https://github.com/btsimonh/hoverboard-firmware-hack/tree/platformio

The output has not been tested on a board yet (I will be a week or so of analysis of existing serial first, then modify for GM32 CPU), so I won't issue a pull request until I'm more sure it will work. 
The .bin file does seem a little larger than the original (~33k, compared to the hex file of ~33k, so about double the size?)

There is one code change, described in the readme.md in the platformio folder. 

If someone wants to try it, and confirm, I have included a .bin file.

lkarlslund, 2018-08-21:
I did just about the same thing a couple of weeks ago, and managed to compile. The binary is way smaller, so I haven't dared flashing it.

btsimonh, 2018-08-21:
hi lars,
both the original and the new work in my GD32 based board.
The binary size seems to vary quite a lot according to config - I think it's bringing in and out bits of the C libraries.  Sometimes a small config change will make a 13k binary, and other times the binary is 30k or more.  The original hex is ~30k, indicating a ~15k binary, but it may skip large gaps.
s

p-h-a-i-l, 2018-08-23:
I have some trouble getting platform.io up and running
`Processing genericSTM32F103RC (platform: ststm32; board: genericSTM32F103RC)

--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: ST STM32 > STM32F103RC (48k RAM. 256k Flash)
SYSTEM: STM32F103RCT6 72MHz 48KB RAM (256KB Flash)
DEBUG: CURRENT(blackmagic) EXTERNAL(blackmagic, jlink, stlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 1 compatible libraries
Scanning dependencies...
No dependencies
Compiling .pioenvs/genericSTM32F103RC/src/startup_stm32f103xe.o
arm-none-eabi-as: unrecognized option '-x'
*** [.pioenvs/genericSTM32F103RC/src/startup_stm32f103xe.o] Error 1
========================== [ERROR] Took 0.64 seconds ==========================`

Also I hade do change your build options to linux compatible paths, but for now i'm stuck. make still works.

btsimonh, 2018-08-25:
ok, pushed a new 'sensorcontrol' branch, and did a pull request against this repo.

p-h-a-i-l, 2018-09-05:
added a commit with case changes and included your changes to bldc.c - now it's compiling. Created pull requests https://github.com/NiklasFauth/hoverboard-firmware-hack/pull/59

zsocietyGithub, 2018-09-18:
i just download file, and try compile in windows 10 using platform io in visual code IDE but not succses compile and error,,,

`Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: ST STM32 > STM32F103RC (48k RAM. 256k Flash)
SYSTEM: STM32F103RCT6 72MHz 48KB RAM (256KB Flash)
DEBUG: CURRENT(blackmagic) EXTERNAL(blackmagic, jlink, stlink)
Warning! Cannot find a linker script for the required board! Firmware will be linked with a default linker script!
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 1 compatible libraries
Scanning dependencies...
No dependencies
Linking .pioenvs\genericSTM32F103RC\firmware.elf
arm-none-eabi-gcc: error: HACK\hoverboard-firmware/STM32F103RCTx_FLASH.ld: No such file or directory
*** [.pioenvs\genericSTM32F103RC\firmware.elf] Error 1`

im new in arm world...

thanks for help

p-h-a-i-l, 2018-09-18:
Which branch/commit did you try?

zsocietyGithub, 2018-09-18:
i use from it  [](https://github.com/btsimonh/hoverboard-firmware-hack/tree/Platform.io)

zsocietyGithub, 2018-09-18:
i using it   :  [https://github.com/btsimonh/hoverboard-firmware-hack/tree/Platform.io](url)

btsimonh, 2018-09-19:
do you have the STM32F103RCTx_FLASH.ld in the root of the repo?
There may be a pathing issue, as it seems to be looking for a relative path 'HACK\hoverboard-firmware/STM32F103RCTx_FLASH.ld' (i.e. not C:\blaa blaa\HACK, nor ./STM32F103RCTx_FLASH.ld
it gets this path from '-Wl,-T${PROJECT_DIR}/STM32F103RCTx_FLASH.ld' in the platform.io ini file.
so maybe ${PROJECT_DIR} is somehow now correct.
Try removing the directory you have added in VSCode, and adding the root of the project instead...
(i.e. your 'hoverboard-firmware' folder).  I'm no Project.io guru :), but this seems a possible....

p-h-a-i-l, 2018-09-19:
Can you try https://github.com/p-h-a-i-l/hoverboard-firmware-hack/tree/pio ?

zsocietyGithub, 2018-09-19:
wait,,,i will try it,,,,

thanks..

btsimonh, 2018-09-19:
ok, it builds for me.  but.  To have PlatformIO recognise it properly, I had to:
1/ ensure you add the specific folder which contains the platformio.ini file to the workspace.
2/ does not like to be added to another workspace which does not already have platformio folders in it? (e.g. added to a php workspace, platformio did not add to the bottom of vscode the build options.)
3/ it adds some josn files in .vscode.
4/ I do get: 'Warning! Cannot find a linker script for the required board! Firmware will be linked with a default linker script!', but it builds.  And I THINK the line '-Wl,-T${PROJECT_DIR}/STM32F103RCTx_FLASH.ld' overrides the platformio chosen script anyway...  (I have some special source in mine, and it works).
5/ Maybe try saving workspace to your project folder....

zsocietyGithub, 2018-09-21:
> ok, it builds for me. but. To have PlatformIO recognise it properly, I had to:
> 1/ ensure you add the specific folder which contains the platformio.ini file to the workspace.
> 2/ does not like to be [...]

today i just try again to compile from your source : [https://github.com/btsimonh/hoverboard-firmware-hack.git](url)
my step is  (correct me if i use wrong methode) : download zip file -> extract file hoverboard-firmware-hack-Platform.io ->open folder from VS code -> compile with platfom.io -> file not compile and get error : 

> Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: ST STM32 > STM32F103RC (48k RAM. 256k Flash)
SYSTEM: STM32F103RCT6 72MHz 48KB RAM (256KB Flash)
DEBUG: CURRENT(blackmagic) EXTERNAL(blackmagic, jlink, stlink)
Warning! Cannot find a linker script for the required board! Firmware will be linked with a
default linker script!
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 2 compatible libraries
Scanning dependencies...
No dependencies
Linking .pioenvs\genericSTM32F103RC\firmware.elf
.pioenvs\genericSTM32F103RC\src\bldc.o: In function `DMA1_Channel1_IRQHandler':
bldc.c:(.text.DMA1_Channel1_IRQHandler+0x250): undefined reference to `blockPhaseCurrent'
bldc.c:(.text.DMA1_Channel1_IRQHandler+0x2a2): undefined reference to `blockPWM'
bldc.c:(.text.DMA1_Channel1_IRQHandler+0x2b2): undefined reference to `blockPWM'
bldc.c:(.text.DMA1_Channel1_IRQHandler+0x2d8): undefined reference to `blockPWM'
bldc.c:(.text.DMA1_Channel1_IRQHandler+0x31a): undefined reference to `blockPWM'
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\genericSTM32F103RC\firmware.elf] Error 1
================================ [ERROR] Took 3.75 seconds ================================

And i use some methode to compile from this source : [https://github.com/p-h-a-i-l/hoverboard-firmware-hack/tree/pio](url)

its compile without error : 

> Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: ST STM32 > STM32F103RC (48k RAM. 256k Flash)
SYSTEM: STM32F103RCT6 72MHz 48KB RAM (256KB Flash)
DEBUG: CURRENT(blackmagic) EXTERNAL(blackmagic, jlink, stlink)
Warning! Cannot find a linker script for the required board! Firmware will be linked with a default linker script!
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 2 compatible libraries
Scanning dependencies...
No dependencies
Checking size .pioenvs\genericSTM32F103RC\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [          ]   3.9% (used 1932 bytes from 49152 bytes)
PROGRAM: [=         ]   7.0% (used 18328 bytes from 262144 bytes)

Thanks all for your guidance 👍

btsimonh, 2018-09-23:
Away at the moment, but for my repo, make sure you are on the pidcontrol branch....

btsimonh, 2018-12-11:
Note new pull request by @p-h-a-i-l :
https://github.com/NiklasFauth/hoverboard-firmware-hack/pull/72
we've been building in platform.io exclusively for some time, and it's fine.

closed 2018-12-11 by btsimonh