lucysrausch/hoverboard-firmware-hack #44: New style board; poweroff not working

btsimonh, 2018-08-19:
Hi All, 
I'm applying this to a YST-6.5"-10"-PCBA-10 board.

The original firmware loads, as does a version I built with platform.io.

However, it only works after a full power off.  A button-press power off stops the motors, and then after a period starts to issue short bleeps (no power down sequence bleep as the code hints at).

Anyone else had this kind of issue?

s

p-h-a-i-l, 2018-08-20:
can confirm, when a motor is spinning, the power button shuts down the motor. Afterwards the board seems stuck until power cycling. Haven't noticed any beeps after pressing.
Shutdown works as expected when motors are stopped.

goosst, 2018-08-23:
can confirm the issue, however I haven't spend time debugging the issue

closed 2018-08-25 by btsimonh

p-h-a-i-l, 2018-08-26:
I was testing with binaries created with make and without attached stlink.