lucysrausch/hoverboard-firmware-hack #48: What software do you guys use!??

SpacePokemon, 2018-08-22:
First of all, thank you guys for your work!

I am new to embedded and I was wondering what software you guys use to program.

I tried to use Atollic TrueSTUDIO but I am getting thousands of errors which I am having hard time to resolve.

Thanks in advance!!

drbytes, 2018-08-22:
I use Keil, it talks to the STLINKv2 and compiles and flashes (called download) the board.

LeoDJ, 2018-08-22:
Normally, on Linux, I use VS Code as an editor and just the plain `make` with `arm-none-eabi-gcc` installed.

Even on Windows, I use the Linux subsystem and compile it using `make` there too.

Just make sure you don't run on gcc version 6, because nothing will compile correctly (you will get a binary, but the board will do weird things or nothing at all, better to not try it). Either verison 4 or 7 is tested to work okay.

I for my part installed gcc 7 using from the team-gcc-arm-embedded PPA like this, because my Ubuntu 18.04 came with the incompatible version 6:
```
sudo add-apt-repository ppa:team-gcc-arm-embedded/ppa 
sudo apt update 
sudo apt install gcc-arm-embedded 
```

There are currently efforts of compiling this project with platform.io too, so I'll probably use that in the future. But for now, `make` works fine enough for me.

PS: as far as I am aware, Niklas is using just plain `make` too, so that's what this project is optimized for.

closed 2018-08-22 by LeoDJ

SpacePokemon, 2018-08-23:
Thank you all!!!
I will try them out and find the one that fits me!!

btsimonh, 2018-08-25:
platform.io also works; look at the latest pull request.  It's really easy to use (on windows) - does pretty much everything for you; although debugging alludes me as yet.

SpacePokemon, 2018-08-29:
@btsimonh 
I just gotta say, platform.io is awesome!!!
Btw did you have any luck debugging with this?