lucysrausch/hoverboard-firmware-hack #49: Question ?

stevenerika, 2018-08-23:
Hello , i am now on a project to  improve the standart hoverboard to use on y company , the thing s that make me so confuse are, the sensor output, i wonder what is the sensor output ? maybe you can help me, thnaks...

sorry for bad english.

LeoDJ, 2018-08-23:
If I understand you correctly, you want to take a normal hoverboard, like for standing on it and do some improvements on that?

If that is the case, I have to disappoint you. This firmware is for controlling the hoverboard motors manually, to make them usable as motors in custom vehicles or robots. No balancing functionality is currently implemented.

PS: As for the protocol that the gyro daughterboards speak to the mainboard, take a look [here](http://drewspewsmuse.blogspot.com/2016/06/how-i-hacked-self-balancing-scooter.html). If that is, what your second question was.

If I understood you incorrectly, please clarify.

closed 2018-08-23 by LeoDJ

stevenerika, 2018-08-24:
i get it... thanks leo for answering...