lucysrausch/hoverboard-firmware-hack #50: Battery / Battery Charger

blubbi321, 2018-08-24:
Hey, 

just a few questions as I could not figure it out from the schematics:

Is it safe to attach a 42V output charger to the +/- BATT ports here? (As the battery is 36V)

[image]

Also, I got a used battery and the output voltage is ~0,5V .. did the guy send me a dead one or is there some circuit in between that just doesnt show the full voltage?

Thanks for any pointers!

LeoDJ, 2018-08-24:
If this charger is in fact a charger and not just a 42V Power Supply (it should say so on its label), then it is fine. Because a fully charged Li-Ion cell has between 4.1-4.2V and the Hoverboard batteries are 10S packs, so 41-42V fully charged.

Regarding your battery. Yes, there is a protection circuit in there and it will shut down, when a cell voltage get's too low. You can try to charge the battery for a few minutes and then measure again. If the voltage is still unter 30V, then your battery might be dead, or at least have a pair of dead cells in it.

blubbi321, 2018-08-24:
Thanks for your reply!

Would that mean that I could directly attach the charger (if its a real charger) to the XT60 connector of the battery (because of the extra circuit integrated in the battery)? Or is the mainboard needed inbetween in any case?

blubbi321, 2018-08-24:
Ok dumb question, directly connected, nevermind ;)

blubbi321, 2019-05-30:
> Regarding your battery. Yes, there is a protection circuit in there and it will shut down, when a cell voltage get's too low. You can try to charge the battery for a few minutes and then measure [...]

Do you know by any chance which other functionality the battery circuit (the one inside the battery wrapping) provides? Does it protect from overcharging for example?

Mathew3000, 2019-05-31:
It is a complete BMS (Battery Management System) which is used to protect each cell (in this case two cells in parallel) against over/undervoltage and over/undercurrent.

Mathew3000, 2019-05-31:
Oh and it handles cell-balancing

Mathew3000, 2019-05-31:
To the power supply, most of them are limited at 42v but limited amperage, which means it is basically a "cheap" implementation of a CC-CV charging.

blubbi321, 2019-06-03:
Thanks for the information, that was what I was hoping for :)