lucysrausch/hoverboard-firmware-hack #53: [FeatureRequest] PID position control

EmerickH, 2018-08-26:
Hello,
Is it possible to implement PID in the motor controller to obtain more accurate speed and position control instead of speed?
I can help you on some parts but I don't know how to do, especially for the position.

Thank you

btsimonh, 2018-08-27:
I've got a branch here https://github.com/btsimonh/hoverboard-firmware-hack/tree/withflashcalibration which includes reading the absolute hall sensor position.  I intend to enhance it to give absolute speed and position, and finish off my serial protocol for robust control.
Separating out the hall position should be quite easy, it's all in one file apart from a single change in Setup.c.  Getting speed should be a matter of reading a timer in the hall interrupt, and so knowing how long the hall segment took to progress.

btsimonh, 2018-08-29:
now enhanced with speed.  Just need to rework the motor drive.  The hallinterrupt.h/c should be pretty portable into any version, just needs a couple of extra bits in _it.c and setup.c.

closed 2020-03-20 by Jan--Henrik