lucysrausch/hoverboard-firmware-hack #71: Motors will not run

patrickomatik, 2018-12-07:
I am working with a pair of recent hoverboards. The PCBs appear visually identical to the photos, but carry the marking: RJ-15105A 01 R1.1 and E252800 RoHS. (Just above is a mirrored legend 0-48V ie. it reads 84V-0 :) )
The firmware builds and flashes, and I get the startup (and power down) melodies. The debug port dumps sensible serial data stream. But whatever I do, I can't get the motors to turn! My crude effort, hardcoding SpeedR=SpeedL=500 in main.c just after the mixer shows up in the serial debug stream, but the motors do not turn.
Is it likely my boards have a slightly different layout, or am I just an idiot and missing something obvious? (It would be really useful to have some basic "known good" hardware tests available for flashing).
Great project, thanks for all the hard work.

p-h-a-i-l, 2018-12-08:
have you checked what the status of enable and timeout are?
I'd try hardcoding
```
enable=1;
timeout=0;
```
too.
what kind of input are you using?

Jabbbi, 2019-04-08:
We had the same problem. Now the wheels spin again.

Only use this with a bit of caution. It will stop any input protection. In our case we have a problem with our inputs - so we have to check our inputs.

closed 2020-03-20 by Jan--Henrik