lucysrausch/hoverboard-firmware-hack #74: IDE debugging problem

Rudolf10000, 2019-01-24:
I try to follow this project , but i am new to Github and to mcu programming.
I have Atollic installed and i can load up the hex file to stm32- F103RB.
But when i try to debug the untouched programm, i get missing file errors.
Is there are better way to compile-debug then atollic ?
Sorry for such a simple question, but i am looking since days for a solution.

Jan--Henrik, 2019-01-29:
Hi, we use Ozone from Segger for debugging.

p-h-a-i-l, 2019-01-29:
Hi, 
I did some debugging with platformio and the stlink. 
Watch out, debugging in platformio is a premium feature, only one month is free. Also the stlink is somewhat limited. I can do a writeup of my workarounds if somebody is interested.

Rudolf10000, 2019-01-29:
Hi PHAIL, Hi Jan Henrik.
Thanks a lot for your interest. 
And yes, PHAIL, I am very interested in your workarounds. I have a little 
experience in programming, but I can't debug the source code.