lucysrausch/hoverboard-firmware-hack #75: UART Control Scheme and acceleration

amarotica, 2019-01-29:
Hi there,
First off, your firmware is awesome! I have tried controlling using ADC inputs but I am wanting to just use UART control because ADC is only single directional. My question is what range of values do I send to the controller to control? I understand I have to use:
  Serial.write((uint8_t *) &turn, sizeof(turn)); 
  Serial.write((uint8_t *) &velocity, sizeof(velocity)); 
  delay(10);
When controlling with arduino 3.3v but I do not understand what the "turn" and "velocity" values should be. Are they int's or bytes or char's or what? Also, what is the range of these values? Is it like -20 to 20 or -10000 to 10000? I cannot find in the config.h or in the other files I've read through what sort of values to send to the control board. Also, other than the bobbycar version, are there any acceleration settings built in or do I need to send increasing velocity values over a certain time frame, calculated by my arduino? 

I really appreciate any input or info about the UART control input setup that anyone can give because currently I feel lost. Thank you

wholepair, 2019-01-29:
I too am working on getting control of the driver via UART for a differential bot. A write-up, a ino, or samples of python script would be much appreciated! 

I bought the hoverboard-breakout board here:
https://github.com/Jan--Henrik/hoverboard-breakout/tree/master/bobbycar_breakout
I don't fully understand the master/slave interconnects but was thinking that would become clear when I populated the pcb w/ components.

Thanks!

Jan--Henrik, 2019-01-29:
Hey @wholepair ,

The interconnect matrix can be used for UART board to board communication. You can just connect RX to TX and TX to RX between to mainboards. But we've never used that feature, therefore we didn't write code for that. (Also I don't really know if somebody else did it)

p-h-a-i-l, 2019-02-04:
Please be careful with the default UART implementation.  There is no feature to validate the input, you can get really bad speed values when there are disturbances on the RX line (which are to be expected, in noisy motor control).
I am using a uart protocol which is under development right now. (have look at https://github.com/NiklasFauth/hoverboard-firmware-hack/pull/72 for further information).

As for the values -1000 to 1000: This is the duty cycle for the motors. -1000 represents 100.0% duty cycle (full speed) backwards.

Combinacijus, 2019-07-24:
So I took some time and modified serial communication code. As far from all of my tests it's smooth even for poorly connected data lines. Motors stops if data lines are disconnected You can find the code in my fork here:
https://github.com/Combinacijus/hoverboard-firmware-hack
There is also example code for Arduino which sends command via TX pin

Combinacijus, 2019-07-24:
So I took some time and modified serial communication code on original NiklasFauth respository. As far from all of my tests it's smooth even for poorly connected data lines. Motors stops if data lines are disconnected. Reconnection also works well. You can find the code in my fork here:
https://github.com/Combinacijus/hoverboard-firmware-hack
There is also example code for Arduino which sends command via TX pin
But original code only have pwm control so you can't just set speed you want well at least it's really smooth