lucysrausch/hoverboard-firmware-hack #76: PPM Optimized Firmware? - Request

snowkingcro, 2019-01-29:
Hi to all, specially Niklas and Jan Henrik - You guys are awesome.
I'm building a project for a year now, and the last month i ran the first time on the GPN18 Video on YT.
I was amazed how easy all this looks, so i downloaded the repo, flashed my Hoverboard, but when I connect my Rc Receiver (2,4Ghz 3ch), nothing happens.
As i'm a noob, i had a hard time to find out how to modify the config.h (and makefile...), but even then, i didn't manage to recompile the Firmware.
Followed a few tuts, even with all old and new toolchains, i admit, i don't know how to do it.
I wonder if anyone could, Please, post the firmware here - like the bobycar-optimized firmware?

Jan--Henrik, 2019-01-29:
Posting a precompiled firmware for every possible kind of control input is something that we can't really accomplish. That's because there is a very huge number of possible controls with a even huger number of combinations (inverted motors, inverted steering, switched channels). And even if we would cover that, then the coefficients for speed and steering are not set. Sorry about that. Could you tell us what you have tried and what the errors are you faced? Then we can find a solution together :)

Jan--Henrik, 2019-01-29:
Oh and thank you for the nice words! :) What kind of project/moving object are you building?

snowkingcro, 2019-01-29:
Well, you are real geniuses who made a huge effort and gave it for free... That's te least i can say.
I'm trying to make a rc controlled  lawnmover.
I first started with dc motors from car wipers, but recently saw a hoverboard laying around, so i came up with building just few updates on my frameto mount the motors, it will look something like your armchair, and as soon as i saw that the board has a ppm input, BINGO!
So, the hoverboard has 10" wheels, and the speed should be around 4-5km/h, but that can be configured on my rc remote.
The main problem is the error 127 that comes up, when i hit "make"
I found out (maybe) that this is because the file was compiled on a 32bit machine, and i am trying to recompile with 64bit Ubuntu.
The makefile confuses me too, because of the path of the prefix... As i said, i am realy a noob in this area...but i would be greatfull for any help! 😉

snowkingcro, 2019-01-30:
Here's a little update, i got back to Windows, pasted the "arm-none-eaby" toolchain folder into the Hoverboard-hack folder, and compiled with cygwin - i think i suceeded. 
[image]
is this something i should get?
is the "hover.hex" the file i should flash?
Thanks!

Jan--Henrik, 2019-01-30:
That looks good to me, yep that is the right file.

snowkingcro, 2019-01-31:
Well, i didn't know that. Thanks for the heads up! As i was just up to flashing my board, guess than this https://goo.gl/images/sygfb9 it's going to do nothing at all...

snowkingcro, 2019-02-01:
UPDATE (Again)
So, thanks to @benjaf for warning about the Rc Car remote and PPM, i ordered a PPM Encoder.
While waiting, i flashed the Niklas's and @Jan--Henrik 's FW, hooked up two 10k potentiometers
and played around.
The first time i powered the Hoverboard up, left wheel was turning, the right stood still, so i lowered the ADC to 50% in order to prevent the wheel to go to max.
Well, i've got some strange results, recorded them for you to see what i got.
[Two potis on ADC input of the Hoverboard](https://www.youtube.com/watch?v=2RZFxEcP9mA)

And here's the code with changed values.

`// ###### CONTROL VIA TWO POTENTIOMETERS ######
// ADC-calibration to cover the full poti-range: connect potis to left sensor board cable (0 to 3.3V) (do NOT use the red 15V wire in the cable!). see <How to calibrate>. turn the potis to minimum position, write value 1 to ADC1_MIN and value 2 to ADC2_MIN. turn to maximum position and repeat it for ADC?_MAX. make, flash and test it.

#define CONTROL_ADC                 // use ADC as input. disable DEBUG_SERIAL_USART2!
#define ADC1_MIN 0                // min ADC1-value while poti at minimum-position (0 - 4095)
#define ADC1_MAX **2048**               // max ADC1-value while poti at maximum-position (0 - 4095)
#define ADC2_MIN 0                // min ADC2-value while poti at minimum-position (0 - 4095)
#define ADC2_MAX **2048**               // max ADC2-value while poti at maximum-position (0 - 4095)

`

I will try to NOT go offtopic now, but, my arduino is already programed to proces The PWM of the RC Receiver, my question is: Would it be possible to program arduinos output to send voltage to ADC?
Then i don't have to mess with PPM encoding and editing the firmware in that way, just only the ADC configuration? Seems to be easyer to me (as a noob).

-

benjaf, 2019-02-03:
You can configure the Arduino to output PWM (No, not the RC sort of PWM) which could be smoothed with a low-pass filter to become an analogue signal.
If you REALLY want to try the board with the RC radio before the PPM encoder shows up, I have a fork of this project containing very basic support for RC PWM input. It is still experimental, and will not support steering. But it may get you started.

snowkingcro, 2019-02-04:
@benjaf, that's perfect.
I flashed your FW, it's amazing how good and smooth the throttle is! And it's working on PWM signal! 
That's what i was actualy searching for, now, the only thing that wonders me, how and where to connect the 2nd chanel for steering? As for steering, i would only need one wheel to turn off. Could the adc1 do this? Just asking...

benjaf, 2019-02-04:
@snowkingcro glad it helped!
I plan to eventually adapt some of the code from the bbcar fork, it has very smooth throttle response and the option to 'boost' top speed quite a bit.
As for adding the steering channel: You'd need to copy and rename several sections of the PWM code if you want to use the transmitter. How difficult this is depends mostly on how comfortable you are with programming.
If you want to use ADC, you pretty much just have to include parts of the ADC code in main.c right around line 221 with a few modifications.

snowkingcro, 2019-02-04:
Well, if i'd know what to paste and where, i would have no problem to recompile and test, and eventualy to make adjustments 😎
Just one thing that wonders me, in that case - should the ch1 and ch2 be jumpered on the receiver to go with only PPM input?

Dan9287, 2019-02-05:
> just click on benjafs Avatar, or name, it get's you directly to his github
> site, and there it is. :)

Idk why that was so difficult for me lol thank you

snowkingcro, 2019-02-05:
I was asuming You are going to do something like this, but sorry, it's not working. :-(
Speed is still working, but steer not. i tried to make some combinations and changes on the receiver (didn't touch the code),  but nothing helps...
below are some pics how i wired it, think therse's nothing wrong... should i maybe play with speed values in config?
[image]
[image]

Dan9287, 2019-02-06:
I AM doing something similar but i dont need steering as i am using two
motherboards for a skid-steer style vehicle. So as long as there is
forward, brake, and reverse, my FrSky Qx7 transmitter can easily do the
channel mixing on its own. I'll have 16ch at my disposal.

benjaf, 2019-02-06:
@snowkingcro The wiring looks fine to me.
It was rather erratic on my board too, sometimes working and sometimes not. I'm not sure if the timer I used for the steering channel might be in use somewhere else as well.
Try disabling DEBUG_SERIAL_USART3 in config.h and see if that helps?

Dan9287, 2019-02-06:
i was able to get both pwm(not including steering) and nunchuck control
working decently, i just need to revers one motor or the other and i am
unsure of how to do that in the code. ive done what i thought made sense
and no dice.

benjaf, 2019-02-06:
@Dan9287 Setting SPEED_COEFFICIENT to -1 should reverse the direction

p-h-a-i-l, 2019-02-06:
I think the proper way to invert is deactivating or activating these settings:
https://github.com/NiklasFauth/hoverboard-firmware-hack/blob/9beae76952119e3bb3e367d43a307f4ac420c6e4/Inc/config.h#L90-L91

snowkingcro, 2019-02-06:
I'm struggling half of the day to recompile after disabling DEBUG_SERIAL_USART3... getting ERROR 139, as i saw that you are using the sam toolchain like me, im now completly lost... i don't know how cygwin works on the master repo, and on your fork not... 😩😩
I guess, I'll just wait for my PPM decoder to arrive.
I realy appreciate all work that @benjaf done, and man, thank you very much!

Dan9287, 2019-02-06:
@snowkingcro/benjaf you guys have both helped me out immensely. Ive nailed
down most of MY issues with disabling steering and getting both motors to
behave the same, what ive run into now is, actual controls are backwards.
Pushing forward on the nunchuck results in reverse movement along with the
"backup beeps". Is there a way to invert control?
Again MAJOR thanks to both of you!

Dan9287, 2019-02-07:
Got it sorted. Thanks again for the input!

p-h-a-i-l, 2019-02-07:
what did you change?

Dan9287, 2019-02-07:
@p-h-a-i-l im trying to figure out what i did to reverse the nunchuck
control, im not entirely sure. i was very sleepy and just trying things til
something worked. here are my main, and config files so you can have a
look. if you can figure out what i did please let me know lol

/*
* This file is part of the hoverboard-firmware-hack project.
*
* Copyright (C) 2017-2018 Rene Hopf <renehopf@mac.com>
* Copyright (C) 2017-2018 Nico Stute <crinq@crinq.de>
* Copyright (C) 2017-2018 Niklas Fauth <niklas.fauth@kit.fail>
*
* This program is free software: you can redistribute it and/or modify
* it under the terms of the GNU General Public License as published by
* the Free Software Foundation, either version 3 of the License, or
* (at your option) any later version.
*
* This program is distributed in the hope that it will be useful,
* but WITHOUT ANY WARRANTY; without even the implied warranty of
* MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
* GNU General Public License for more details.
*
* You should have received a copy of the GNU General Public License
* along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/

#include "stm32f1xx_hal.h"
#include "defines.h"
#include "setup.h"
#include "config.h"
//#include "hd44780.h"

void SystemClock_Config(void);

extern TIM_HandleTypeDef htim_left;
extern TIM_HandleTypeDef htim_right;
extern ADC_HandleTypeDef hadc1;
extern ADC_HandleTypeDef hadc2;
extern volatile adc_buf_t adc_buffer;
//LCD_PCF8574_HandleTypeDef lcd;
extern I2C_HandleTypeDef hi2c2;
extern UART_HandleTypeDef huart2;

int cmd1;  // normalized input values. -1000 to 1000
int cmd2;
int cmd3;

typedef struct{
   int16_t steer;
   int16_t speed;
   //uint32_t crc;
} Serialcommand;

volatile Serialcommand command;

uint8_t button1, button2;

int steer; // global variable for steering. -1000 to 1000
int speed; // global variable for speed. -1000 to 1000

extern volatile int pwml;  // global variable for pwm left. -1000 to 1000
extern volatile int pwmr;  // global variable for pwm right. -1000 to 1000
extern volatile int weakl; // global variable for field weakening left. -1000 to 1000
extern volatile int weakr; // global variable for field weakening right. -1000 to 1000

extern uint8_t buzzerFreq;    // global variable for the buzzer pitch. can be 1, 2, 3, 4, 5, 6, 7...
extern uint8_t buzzerPattern; // global variable for the buzzer pattern. can be 1, 2, 3, 4, 5, 6, 7...

extern uint8_t enable; // global variable for motor enable

[... 412 lines cut ...]
else {\
  weakl = 0;\
  weakr = 0;

// ############################### VALIDATE SETTINGS ###############################

#if defined CONTROL_SERIAL_USART2 && defined CONTROL_ADC
  #error CONTROL_ADC and CONTROL_SERIAL_USART2 not allowed. it is on the same cable.
#endif

#if defined CONTROL_SERIAL_USART2 && defined CONTROL_PPM
  #error CONTROL_PPM and CONTROL_SERIAL_USART2 not allowed. it is on the same cable.
#endif

#if defined DEBUG_SERIAL_USART3 && defined CONTROL_NUNCHUCK
  #error CONTROL_NUNCHUCK and DEBUG_SERIAL_USART3 not allowed. it is on the same cable.
#endif

#if defined DEBUG_SERIAL_USART3 && defined DEBUG_I2C_LCD
  #error DEBUG_I2C_LCD and DEBUG_SERIAL_USART3 not allowed. it is on the same cable.
#endif

#if defined CONTROL_PPM && defined CONTROL_ADC && defined CONTROL_NUNCHUCK || defined CONTROL_PPM && defined CONTROL_ADC || defined CONTROL_ADC && defined CONTROL_NUNCHUCK || defined CONTROL_PPM && defined CONTROL_NUNCHUCK
  #error only 1 input method allowed. use CONTROL_PPM or CONTROL_ADC or CONTROL_NUNCHUCK.
#endif

p-h-a-i-l, 2019-02-07:
;) I was just asking for future googlers. It's always annoying to find old postings describing your problem without a solution.

p-h-a-i-l, 2019-02-07:
ok, here is what you did :)

in main.c you reversed pwmr (could have done the same by removing `#define INVERT_R_DIRECTION` from config.h
it would look like this:
```
//#define INVERT_R_DIRECTION
#define INVERT_L_DIRECTION
```
(the other changes to your main.c do not really do anything :) )

your speed coefficient is negative in config.c
so this is again changing the state of both INVERT_R_DIRECTION and INVERT_L_DIRECTION.

so your config would now be like:
```
[...]
//#define CONTROL_ADC                 // use ADC as input. disable DEBUG_SERIAL_USART2!
[...]
#define CONTROL_NUNCHUCK            // use nunchuck as input. disable DEBUG_SERIAL_USART3!
[...]
#define FILTER              1.0  // lower value == softer filter. do not use values <0.01, you will get float precision issues.
#define SPEED_COEFFICIENT   1.8  // higher value == stronger. 0.0 to ~2.0?
#define STEER_COEFFICIENT   0.0  // higher value == stronger. if you do not want any steering, set it to 0.0; 0.0 to 1.0
#define INVERT_R_DIRECTION
//#define INVERT_L_DIRECTION
#define BEEPS_BACKWARD 1    // 0 or 1
[...]
```
with a clean repo you should get the same behaviour

Dan9287, 2019-02-07:
when i commented "invert_" on either motor i always got clock skew errors,
or it had no effect. but then again im literally stumbling through this not
having REAL suitable experience so idk lol my thing was that when pushing
forward on the nunchuck, motors were in reverse, and i got reverse beeps.
the actual motor direction wasnt the main issue. so i would have had to
hold the nunchuck upside down to have proper control. i am running one
board per side(one left, one right) on my "skid-steer" so i had to get both
motors (two per side) to respond correctly when in the same orientation on
my vehicle. i will be controlling the boards with nunchucks independently
of eachother, so that i am responsible for steering mix with my input.
my current issue is that no matter how i set the filters, i am getting
slight forward input to the board with nunchuck in resting state. is there
some sort of way to set a deadband to mitigate the erroneous input while in
resting state? or is this issue something that could be solved by an i2c
pullup or pulldown?

p-h-a-i-l, 2019-02-07:
backward beeping is based on "speed", which is only based on input. So it shouldn't beep. But anyways, you could also just disable backwards beeping..
FILTER only does smoothing of the speed and steer, it does not set an offset. 
You can either program a way to calibrate the nunchuck, or hard code an offset.
Just replace 127 and 128 with values fitting for your nunchuk
https://github.com/NiklasFauth/hoverboard-firmware-hack/blob/9beae76952119e3bb3e367d43a307f4ac420c6e4/Src/main.c#L188-L189

snowkingcro, 2019-02-08:
My PPM decoder seems to be delayed in shipping because the Chinese New Year Holidays... :/
So i have no choice but to get my compiler working again.
@benjaf i disabled UART3, but i got this message...any toughts?
[image]

benjaf, 2019-02-08:
@snowkingcro That seems very strange. I tried disabling DEBUG_SERIAL_USART3, and the build still works just fine. I don't know much about Cygwin though, I've been using MinGW. Looks like it's not able to find/read some of the files?

p-h-a-i-l, 2019-02-08:
I found it very easy to get a new system to compile when using platformio. It's usually three steps: 
1. install vscode
2. add platformio extension
3. clone git repository

snowkingcro, 2019-02-08:
@p-h-a-i-l i have visual c instaled, but platformio seems not working for me... I don't have the ability to build (or compile)...
I reinstaled, tried a few things, but i use it to edit the code. 
@benjaf, I'll try the miniGW, but for the missing files, i don't know why i get this... Would it help if i only copy config.h into the original repo? Or should i copy something else?

p-h-a-i-l, 2019-02-08:
@snowkingcro The code needs some small modifications to compile in platformio (and a .ini file is necessary too)
A working example can be found here https://github.com/p-h-a-i-l/hoverboard-firmware-hack/tree/pull

benjaf, 2019-02-08:
@snowkingcro A clean repository often fixes that sort of problems. Try just commenting that one line in a new clone of the repo, and copy your own make file too - mine probably won't work for you.

Dan9287, 2019-02-09:
Hello again gents, ive created a simple facebook group to help keep some of
the clutter out of the "issues" . please feel free to join, i am the sole
admin (currently) but i will need some help for the real brains behind
these projects.

"Hover-Hacks"
https://www.facebook.com/groups/285402365465113/

Anyone willing to help, and share knowledge in the group will be greatly
appreciated.
 The group is only 5min old so the page is pretty barren at the moment.
Please share and invite also! Thank you!

snowkingcro, 2019-02-11:
@benjaf i followed your advice, cleaned the repo, used original makefile, disabled UART3 finaly i managed to compile... but without success, nothing seems to work now... flashed again with your hex, so i got the speed again... 😊

Jan--Henrik, 2019-02-12:
@Dan9287 just that you know, we already do have a [Telegram Group](https://t.me/joinchat/BHWO_RKu2LT5ZxEkvUB8uw) with ~60 members.

copterfight, 2019-03-06:
> @benjaf i followed your advice, cleaned the repo, used original makefile, disabled UART3 finaly i managed to compile... but without success, nothing seems to work now... flashed again with your hex, so i got the speed again...

i build @benjaf firmware, it working but with bug, https://youtu.be/4pRSbMTv8mA

snowkingcro, 2019-03-06:
This is not a bug. I had the same problem, cut the wires shorter!
This solved my problem.

Jan--Henrik, 2019-03-07:
Motor control (in general) can introduce quite some noise, you should suppress that by adding filters to the  data lines. For example: ferrite bands or capacitors

BorgStefan, 2019-03-31:
> @snowkingcro A clean repository often fixes that sort of problems. Try just commenting that one line in a new clone of the repo, and copy your own make file too - mine probably won't work for you.

> The PPM code is unlikely to work for you, if you have a standard 3 channel RC receiver. What is known as PPM in RC is generally only available on high-end 8 channel systems, mainly targeted at drones. It compresses all of the channels into a single 3-wire connector rather than having one connector per channel. This is what is supported by the firmware.
> What most common receivers use is known as PWM. The duration of the pulse varies between 1 and 2 milliseconds every ~20 milliseconds, and the timing doesn't really matter. Some changes will be required in order to get it to work! One channel should be relatively straightforward, I don't know about 2 or more. But maybe someone else have tried it?

@benjaf 
Thx Benjaf. I used your code to compile and flash my mainboard from a hoverboard. Then used a RC remote 2,4 Ghz 4 channels and it worked!! Thx for your code and if it would be possible I would show a video on it. Here is a picture at least.

franck24300, 2019-06-09:
@benjaf 
Hello. I can not compile your program, even without modification. When I try I get this error:

franck@DESKTOP-TGVQI7F:/mnt/c/firmwaremodifie/hoverboard-firmware-hack-esk8-master$ make
arm-none-eabi-gcc" -c -mcpu=cortex-m3 -mthumb   -DUSE_HAL_DRIVER -DSTM32F103xE -IInc -IDrivers/STM32F1xx_HAL_Driver/Inc -IDrivers/STM32F1xx_HAL_Driver/Inc/Legacy -IDrivers/CMSIS/Device/ST/STM32F1xx/Include -IDrivers/CMSIS/Include -Og -Wall -fdata-sections -ffunction-sections -std=gnu11 -g -gdwarf-2 -MMD -MP -MF"build/stm32f1xx_hal_flash.d" -MT"build/stm32f1xx_hal_flash.d" -Wa,-a,-ad,-alms=build/stm32f1xx_hal_flash.lst Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_flash.c -o build/stm32f1xx_hal_flash.o
/bin/sh: 1: Syntax error: Unterminated quoted string
Makefile:139: recipe for target 'build/stm32f1xx_hal_flash.o' failed
make: *** [build/stm32f1xx_hal_flash.o] Error 2
franck@DESKTOP-TGVQI7F:/mnt/c/firmwaremodifie/hoverboard-firmware-hack-esk8-master$

BorgStefan, 2019-06-10:
Check in makefile that ”HOME = xxxx”
Is defined correctly.

Br Stefan

franck24300, 2019-06-10:
@BorgStefan 
I defined as in the makefile of Niklas, which I can compile if I do not modify anything : PREFIX = arm-none-eabi-
To compile I use the method described here for Windows 10 : https://github.com/NiklasFauth/hoverboard-firmware-hack/wiki/Build-Instruction:-TranspOtter

BorgStefan, 2019-06-10:
Hi,

I am a bit confused now. In your first mail, to my understanding you state you get the error message with no changes made to the code.
Then now below you state the opposite?

I still believe there is a problem how "Home" is defined in the "Makefile"

I compile in my unix environment.

BR Stefan

franck24300, 2019-06-10:
Sorry for the confusion. If I try to compile the program without any modifications, I get the following error:
$ make
"C:/Program Files (x86)/GNU Tools Arm Embedded/7 2018-q2-update/bin/arm-none-eabi-gcc" -c -mcpu=cortex-m3 -mthumb   -DUSE_HAL_DRIVER -DSTM32F103xE -IInc -IDrivers/STM32F1xx_HAL_Driver/Inc -IDrivers/STM32F1xx_HAL_Driver/Inc/Legacy -IDrivers/CMSIS/Device/ST/STM32F1xx/Include -IDrivers/CMSIS/Include -Og -Wall -fdata-sections -ffunction-sections -std=gnu11 -g -gdwarf-2 -MMD -MP -MF"build/stm32f1xx_hal_flash.d" -MT"build/stm32f1xx_hal_flash.d" -Wa,-a,-ad,-alms=build/stm32f1xx_hal_flash.lst Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_flash.c -o build/stm32f1xx_hal_flash.o
/bin/sh: 1: C:/Program Files (x86)/GNU Tools Arm Embedded/7 2018-q2-update/bin/arm-none-eabi-gcc: not found
Makefile:139: recipe for target 'build/stm32f1xx_hal_flash.o' failed
make: *** [build/stm32f1xx_hal_flash.o] Error 127

If I change in the makefile PREFIX = arm-none-eabi-
I get the error described previously. 
Sorry for my ignorance but I can not find HOME = in the makefile.

BorgStefan, 2019-06-10:
Hi, No the Home is not there. I was wrong 🙂

I will check when I get home to my linux machine.

BR Stefan

franck24300, 2019-06-10:
Finally I solved the problem simply by replacing Benjaf's Makefile with Niklas's Makefile, and now I can edit and compile Benjaf's program.
What should I change in the program of Benjaf so that the wheels are braked permanently when the wheels do not turn, ie when the joystick is in neutral? I modified this line in the config.h, the wheels stop faster but then they are not braked.
#define BRAKE_FILTER        0.9f

benjaf, 2019-06-11:
@snowkingcro If you want to have the wheels permanently powered, you need to remove the following section of main.c (line 282):
`    // Disable motors when throttle is in neutral
    if(cmd2 == 0) {
      enable = 0;
    } else {
      enable = 1;
    }`

franck24300, 2019-06-11:
@benjaf  Ok, thanks, i will try.

franck24300, 2019-06-11:
@benjaf  Is there an option to change somewhere to have the steering functional, because in the current state the steering do not working?

benjaf, 2019-06-11:
@franck24300 I have no steering support in my version. I tried to implement it, but for some reason it didn't work as expected - and since I didn't need steering It was mostly removed.
There is still some of the code in there if anyone should want to give it a try. In theory it should be fairly simple, so I don't know why it caused so much trouble.

snowkingcro, 2019-06-14:
Hi all, was some time busy so my project got halted.
dear @benjaf - i'm very thankful for your effort building a .hex with PWM support. too bad i didn't got steering. :1st_place_medal: 
@p-h-a-i-l  - I tried working on VisualCode, and got Platformio to build the .hex, so i finaly got that sorted too. :+1: 
After countless month's of waiting, my PPM Decoder arrived.
I got it tested on an oscilloscope so from my (2ch RC Car R/T) PWM input i really get  PPM output.
So, after defining PPM input, and successful compiling, i am getting nothing. the board powers on, and nothing is happening :(
@Jan--Henrik  can you please confirm that the PPM input is working?

[image]
[image]
[image]

franck24300, 2019-06-14:
It's the same for me too with a Hitec Optima D PPM receiver and the Niklas firmware, the wheels do not turn.

Jan--Henrik, 2019-06-14:
Hey, PPM Sum input works properly. Make sure that you have a sum signal and not just a plain PPM signal. Also please note, that it only works if you set the number of channels that the receiver puts out, is configured in the firmware.

franck24300, 2019-06-14:
Yes ! That was it. I changed the number of channels on 8, the number of channels of my receiver, and it works. Thank you !

snowkingcro, 2019-06-14:
@franck24300 would you be kind and share your hex with me?

franck24300, 2019-06-14:
Yes here is the file hover.hex

[hover.zip](https://github.com/NiklasFauth/hoverboard-firmware-hack/files/3292256/hover.zip)

franck24300, 2019-06-15:
A little video to show that it works

[https://youtu.be/fROINR5-P9c](url)

uhi2000, 2019-06-18:
perfect, if you can do that with a big anlog Poti Joy stick or Nunchuk  I would be very grateful .... no RC

franck24300, 2019-06-19:
@benjaf  Hello
Is it possible in your program to reverse the direction of rotation of a single wheel? I would like one wheel to rotate in one direction and the other to reverse.

[28 more replies]

closed 2019-07-05 by snowkingcro