lucysrausch/hoverboard-firmware-hack #78: Controller always in Backward mode as soon as i activate one input

FPV-Rover, 2019-02-06:


p-h-a-i-l, 2019-02-06:
I can only answer one part of your question, but maybe it is connected? 
The inability to switch off is an old bugfeature. If your speed (set point from your input) is above 20, the board will not power off. See here:
https://github.com/NiklasFauth/hoverboard-firmware-hack/blob/9beae76952119e3bb3e367d43a307f4ac420c6e4/Src/main.c#L79-L90
But since enable is also set to 0 before poweroff was called, the motors will be deactivated.
https://github.com/NiklasFauth/hoverboard-firmware-hack/blob/9beae76952119e3bb3e367d43a307f4ac420c6e4/Src/main.c#L275-L280

I added a specific beeping and timeout to my code to make the input problem more transparent. It is actually good to make sure that your inputs are 0 when parking. Otherwise you can get nice surprise when you start up again :)

Are you sure that you beeping is backwards beeping? There are also other warnings encoded in beeping.. Did you switch off the backwards beeping feature to test?

FPV-Rover, 2019-02-06:
Thanks for the quick response :-) Yeah the switching off problem was just as additional hint for something doesn´t fit :-D
It´s connected with the ottercontrol board and the oszi shows me signals on the TX wire, so at least that should work.  But no movements.

Yeah i commented the line (line 92 in config.h) with 
`#define BEEPS_BACKWARD 1    // 0 or 1`
to 0 and it stopped beeping after flashing.

p-h-a-i-l, 2019-02-07:
Sorry, I'm not familiar with the ottercontrol board. Is it connected via UART? I'd focus on one input Method and if that's working. Do you use multiple inputs at once, or only one at a time?
It also helps to get the UART debug line up and running, you can monitor the values this way.

Jan--Henrik, 2019-02-07:
@FPV-Rover What is the voltage of your power supply? At lower voltages (I think it was below ~33V) the board can no longer be switched off due to the button circuit.

FPV-Rover, 2019-02-12:
Well :-D i found the reason for the Problem.. i trusted the comments... :-) So can you please change the following lines in the config.h?: (Line 73,74,75)

` ###### CONTROL VIA NINTENDO NUNCHUCK ######
// left sensor board cable. keep cable short, use shielded cable, use ferrits, stabalize voltage in nunchuck, use the right one of the 2 types of nunchucks, add i2c pullups. use original nunchuck. most clones does not work very well.
//#define CONTROL_NUNCHUCK            // use nunchuck as input. disable DEBUG_SERIAL_USART3!`
to
`//###### CONTROL VIA NINTENDO NUNCHUCK ######
// right sensor board cable (PB10, PB11). keep cable short, use shielded cable, use ferrits, stabalize voltage in nunchuck, use the right one of the 2 types of nunchucks, add i2c pullups. use original nunchuck. most clones does not work very well.
//#define CONTROL_NUNCHUCK            // use nunchuck as input. disable DEBUG_SERIAL_USART3!` 

I´ll close this topic.

Thanks for your help :-) 

Moe

closed 2019-02-12 by FPV-Rover