lucysrausch/hoverboard-firmware-hack #80: Original firmware

StefanGas, 2019-02-13:
Is there a way to have a look on the original firmware you started with?
I need the IMU support for the balancing behavior.

Thanks
Stefan

rene-dev, 2019-02-13:
this was coded from scratch, its not based on the original hoverboard firmware.

MTRatcliffe, 2019-02-13:
https://github.com/btsimonh/hoverboard-firmware-hack
Git above is based on this repo, but has the option to use it as a hover board again according to Jan Henrik, for this it must be reading the sensor boards, I haven't tested it yet.  Im also interested in seeing these boards in self balancing mode.

MTRatcliffe, 2019-02-13:
Thinking about this, with a self balancing machine you may have issues with the resolution of the motor. That is the hall sensors offer 90 positions per revolution of the motor, with a 6.5 inch motor/tyre combo this means a forward/backward step of 5.76mm this is quite large for self balancing machine [or at least I think it will be]. 

To get it to work well I think it will need the feedback loop to control the power of the motor not forward/back via steps.

rene-dev, 2019-02-13:
balancing doesn't need any feedback from the motors. it needs a IMU.

MTRatcliffe, 2019-02-13:
Yep you are correct for sensing angle it uses a IMU. Many of the examples online are for stepper based robots but further down in the output of the feedback loop from the IMU these examples are controlling the tyres in terms of motor steps not power. In this case that would be via its change in hall-signals [or at least thats my understanding].

Unless I miss understood the code [highly likely] and its able to switch from forwards to backwards power without waiting for a hall click.

StefanGas, 2019-02-14:
What about the "leaked" software they mention in the gpn18 video.
I just like to optimize the control parameters for such a kind of board:
https://evnerds.com/electric-vehicles/e-unicycle-hoverboard-news/diy-onewheel-electric-skateboard-from-poland/

Jan--Henrik, 2019-02-14:
@StefanGas The leaked firmware was already leaked modified, it will not work as original hoverboard firmware. This firmware is, as @rene-dev mentioned, coded from scratch. The first version, which is deprecated and should not be used anymore, of this firmware is based on the "leak" https://github.com/NiklasFauth/Hoverboard-Board-Hack

rene-dev, 2019-02-14:
the "leaked" software cannot be related to the original firmware. It has bugs that cause fets to explode, and this does not happen with the original software.

closed 2019-06-20 by Jan--Henrik