lucysrausch/hoverboard-firmware-hack #81: How come it can only be powered off when speed < 20?

hoonkai, 2019-02-25:
Hi

Just a quick question: I was looking at the code and noticed this line: https://github.com/NiklasFauth/hoverboard-firmware-hack/blob/master/Src/main.c#L80

where `speed < 20` must be true for the board (?) to switch off. Why 20?

Cheers

larsmm, 2019-02-25:
This is for safety. You would not be able to steer or break if it would power off on high speed. 20 is nearly no speed. Use other values if you want.

hoonkai, 2019-02-25:
@larsmm But as for emergency power off, wouldn't you want to power it off regardless? For instance, when the battery is undercharged.

larsmm, 2019-02-25:
Long before battery is in an undercharged state the warning beeper gets activated. If you ignore that and drive a very long time without stopping, it can get dangarous for your battery. I think crashing uncontrolled into something or someone is worse then a broken battery.

p-h-a-i-l, 2019-02-25:
I think it would be good to move this check out of the poweroff function.
Especially here https://github.com/NiklasFauth/hoverboard-firmware-hack/blob/bf2359ec6d169aae6e16bb8984720e471ac049eb/Src/main.c#L277 it is quite confusing. On pushing the off button, the motors stop (enable = 0), but when the speed set point is still set, the board will not power down.