lucysrausch/hoverboard-firmware-hack #82: How are the 2 mainboards connected in the Bobby car project?

BorgStefan, 2019-03-03:
[image]
Hi Niklas, 

Firstly a very nice bobby car! It inspired me to build this skateboard with 4 ten inch wheels. 
My question to you, if you have time and support it, is how are the 2 mainboard connected or not connected. I have difficulties in understanding how your 4 wheel bobby car is controlled. 

Hope this is okey to ask.

BR Stefan

Jan--Henrik, 2019-03-03:
Hey @BorgStefan,

The two mainboards inputs are connected in parallel, both get the same analog signal. No other communication is used.

BorgStefan, 2019-03-03:
OK, Understood. Will try and flash the mainboard soon.
 I will try and use a remote wireless connected to the mainboard if I can.
Thx for your quick answer!!
BR Stefan

larsmm, 2019-03-05:
Here some information about that: https://larsm.org/allrad-e-bobby-car/

BorgStefan, 2019-03-05:
> Here some information about that: https://larsm.org/allrad-e-bobby-car/

Thx larsmm, but my german is limited. Sorry for that.

BR Stefan

BorgStefan, 2019-03-31:
Hi,

Managed to compile and flash the mainboard. 
I also managed to get my 2.4 Ghz 4 channels remote/receiver to run with the mainboard in excellent way. 
Jan -- Henrik and Lars. wanted to upload a video the skateboard using 2 out of 4 wheels. Goes really fast... if you think 15-20 km/h fast. At least on a skateboard that I built...

BorgStefan, 2019-04-01:
> Here some information about that: https://larsm.org/allrad-e-bobby-car/

Hi Lars,
Could i ask how you solved power on for 2 mainboards. 
I will think I will have 2 power on buttons, but use same powersuply for both. Is this the way you solved it? 
Or do you have 1 power on button for both mainboards?

BR Stefan

larsmm, 2019-04-01:
I have 1 button for 2 mainboards. I connected it via Y cable directly to both mainboards.

andredemari, 2019-08-04:
@BorgStefan I had the same problem, try to invert the power button wires of one of the boards and connect them in a Y cable. It worked for me.