lucysrausch/hoverboard-firmware-hack #83: [Question] ADC control via 2 pots erratic behavior

jabulonboaz, 2019-03-03:
Hi thanks for this brilliant work Jan and Niklas. Basically my issue is when i conect 2 potentiometers to the left sensor RX and TX i get a strange behavior.  Turn the 1st pot and the right wheel trurns one direction then trun that pot back to its original position and turn the 2nd pot at wich point both wheels start to counter-rotatate. Is there a guide on how to set the pot parameters and is this the expected behaviour? I want to be able to turn the wheels like in a tank.
Thanks

linkmaker24, 2019-05-16:
i have the same problem using two pots. is there any available solution for this? Thanks

Jan--Henrik, 2019-05-16:
Hey, that is expected because in standard configuration the ADCs are used for speed and steering.
If you want to change that, please have a look at src/main.c line 227.

closed 2019-06-20 by Jan--Henrik