lucysrausch/hoverboard-firmware-hack #84: What is the syntax of comands to control the motos over UART?

jabulonboaz, 2019-03-11:
Using say microcon or any other serial tool on a pi or Ubuntu machine. What is the syntax of comands to control the motos over UART please?
Forward, back, counter rotate left and right, increasing and decrease speed?

AntumArk, 2019-03-12:
First of all have you even tried reading the code? Especially config file. This question has been answered long time ago.

p-h-a-i-l, 2019-03-12:
The main repo does offer only a very limited serial control method. IMO even dangerous to use. You could use it as a starting point, there is Arduino example code in config.h

There are forks which implement a more advanced serial protocol. See https://github.com/btsimonh/hoverboard-firmware-hack. (Since this fork was crashing for me, I have a light version of the protocol here: https://github.com/p-h-a-i-l/hoverboard-firmware-hack)

All of these options are not fully stable yet, you have to be prepared to dig into the code.

Combinacijus, 2019-07-24:
So I took some time and modified serial communication code. As far from all of my tests it's smooth even for poorly connected data lines. You can find the code in my fork here:
https://github.com/Combinacijus/hoverboard-firmware-hack
There is also example code for Arduino