lucysrausch/hoverboard-firmware-hack #87: Help required on ADC wiring - where to pick up 3.3v

barnyrelph, 2019-04-07:
Firstly, thanks so much for sharing this project! I've picked up a couple of dead hoverboards and have:
 - got the toolchain installed
 - built the code
 - flashed one of my controller boards
 - gotten serial debug output from the board

So far so awesome!
I'm looking to make a small e-bike with my son, so I'm planning on using the BobbyCar fork and driving it from a resistive throttle. Looking at the docs and comments  in config.h, it makes sense that I want a voltage divider to supply 0-3.3v to the ADC. The bit I'm stuck on is where to pick up 3.3v on the controller board. 
The only place I can see 3.3v is on the programming header. Can I use that with a variable resistor to ground on the left sensor board to form a voltage divider? Are any other passives needed in this setup?
Sorry if the question seems daft, I'm better on the software side of things

Thanks

Barny

barnyrelph, 2019-04-08:
Ooooh, I just found instructions in https://github.com/NiklasFauth/hoverboard-firmware-hack/wiki/Build-Instruction:-TranspOtter - looks like picking up 3v3 from the debug header is the way to do it! 
I assume that the other way to do this would be to use the bobbycar breakout board with a 3.3volt regulator installed? I've ordered some PCBs based on the latest gerbers, as that cost next to nothing on one of the PCB ordering sites. 

Thanks again for sharing, very excited to get some wheels turning!

NiklasFauth, 2019-04-08:
Yup, just use the 3V3 on the programming connector.
Some tips: 

- Instead of using a potentiometer as a voltage divider alone, use a discrete resistor as a pulldown and the potentiometer as a pullup to form a voltage divider. This way if the potentiometer or a wire going to it break, the vehicle decelerates automatically

- Add a ~1nF ceramic capacitor (eg. for a 10k potentiometer) between PA2 / PA3 and ground as close to the mainboard as possible. This can help filtering the signal and makes driving more smooth.

Best Niklas

closed 2019-04-08 by NiklasFauth