lucysrausch/hoverboard-firmware-hack #89: Restore hoverboard firmware

gaucho1978, 2019-04-13:
I understood it is BOT possibile to backup original firmware due to memory  lock. After unlock the Arm and playing with this modified firmware, could I flash a firmware that allows me to use again the device as a hoverboard?

sh-hilden, 2019-05-03:
You can try this firmware: https://github.com/bipropellant/hoverboard-firmware. It is a firmware to use the hoverboard as a hoverboard

123JRM, 2019-08-05:
Niklas has made the above mentioned fork available at https://pionierland.de/hoverhack/index.php if your interested you can compile a working .bin directly.

closed 2020-03-20 by Jan--Henrik

FarhatullahKhanbuzdar, 2020-08-05:
dear sh-hilden

> You can try this firmware: https://github.com/bipropellant/hoverboard-firmware. It is a firmware to use the hoverboard as a hoverboard

i cant find downloadable file like hex file in this code to use hoverboard as hoverboard.

weiminshen99, 2023-05-19:
I have saved the original Factory flash from both the Master and Slave board before I hacked these boards.

If anyone is still interested, I believe these saved *.bin files can be flashed back to the boards and make them become the original hoverboard again.

Candas1, 2023-05-20:
> I have saved the original Factory flash from both the Master and Slave board before I hacked these boards.
> 
> If anyone is still interested, I believe these saved *.bin files can be flashed back to the boards and make them become the original hoverboard again.

If you are talking about splitboards, this[https://github.com/RoboDurden/Hoverboard-Firmware-Hack-Gen2.x](this)  repository is currently active, you could share it there.

But there are many different board types so it's important to identify what type you have.