lucysrausch/hoverboard-firmware-hack #91: How to connect the button And i2c Lcd screen?

rojrung, 2019-04-17:
I successfully got my hoverboard control by ADC from joysticks
But now I want to build a Go kart with acceleration foot paddle
As I see in config.h it mentions on “button” for speed up the Bobby car
But I don’t know which wire are use for button on right sensor

And for the wheelchair I want to see speed and battery indicator
So how to connect i2c lcd to right sensor cable

Thank you for all of you for this beautiful work

verial, 2019-08-30:
failed too

benjaf, 2019-08-30:
The i2c LCD code doesn't seem to work, so that is probably a part of the problem. If anyone has a good idea on how to fix it, that would be great!

verial, 2019-08-30:
if you help me, I'll do it.

verial, 2019-09-02:
https://www.youtube.com/watch?v=myu9lDYsH0E&feature=youtu.be

closed 2020-03-20 by Jan--Henrik