lucysrausch/hoverboard-firmware-hack #92: Self balancing, change sensitive etc

PeterStanik, 2019-04-17:
Hey! Great projects!

I made onewheel style project. 
And have a question.
I would like to change sensor sensitive, and a few other thinks.
I wonder, what is better way for me. 
Your software + arduino, or try sensor signal hack by Drew Dibble?
I have no programming skills, arduino is simple, and i think, i can make it.

So... any suggestion?
Sorry for my english... 

[image]

snackfart, 2019-08-15:
maybe this helps you with your decision. https://www.youtube.com/watch?v=pvdaV5xP1SM&t=921s

snackfart, 2019-08-15:
i would say buy a gyro and feed the output to the hoverboard controller.,

Chaos99, 2019-12-10:
I'm doing a similar project and I'm facing the same issue. Sensitivity of the original firmware isn't high enough for this use case. I'd even like to change from "lean angle controls speed" to "lean angle controls acceleration" if possible.

What approach did you end up using?

I'm probably going to try to flash this firmware, recreate the original (hoverboard) settings and functionality as good as possible and start modifying from their.
Any hints much appreciated.

Dragopaj, 2023-09-04:
> I'm doing a similar project and I'm facing the same issue. Sensitivity of the original firmware isn't high enough for this use case. I'd even like to change from "lean angle controls speed" to "lean angle controls acceleration" if possible.
> I'm probably going to try to flash this firmware, recreate the original (hoverboard) settings and functionality as good as possible and start modifying from their. Any hints much appreciated.
 Hey, I am facing the same issue, what did you end up doing?