lucysrausch/hoverboard-firmware-hack #93: Getting Debug Data from USART3?

krebera, 2019-04-23:
Hi there! Firstly just wanted to say kudos on this project. It's really really cool.
I have a board identical to the ones in the provided images. I've successfully flashed the board, and get the proper little melodies on power up and down.

However, I cannot get the board to respond to any serial input. More importantly, I can't even read the serial debug output from PB10/PB11 on the right sensor board. I've attempted this using multiple microcontrollers and am currently using a pi to monitor the UART and print when it gets data. I'm running at 115200 baud. Any ideas? Thanks.

P.S: I've also tried configuring it for ADC mode, which doesn't work. I then tried hard-coding motor speed values into the main code, and the only thing that happens is they twitch for the shortest amount of time on startup.
Lastly, I modified the code to just emit a beep on start so I could verify I was actually pushing updates to the board, and the beep worked, but only for a few seconds.

IDK if that information is helpful at all. I'd really appreciate some assistance with this. I'm very exited about this project.

p-h-a-i-l, 2019-04-23:
Does shutdown work? Can you hear the shutdown melody?

krebera, 2019-04-23:
Hey, now that you mention it, the shutdown melody actually doesn't occur! Might there be another reason that's not happening?

I had already removed any modifications I had made to main.c. It's in the state it was in when I originally cloned the repo which was around a week ago.

p-h-a-i-l, 2019-04-23:
change this line
https://github.com/NiklasFauth/hoverboard-firmware-hack/blob/f06ce1fcd33c2e9238f2276858ac639fc2b03674/Src/main.c#L80
to `if(1) {`
and check if shutdown still does not play the melody. If not check if 12V (on my boards 15V) on the Sensor wires is still active. You can also measure if 3.3V is active on your board. 
I think you might have a hardware issue with your board.

krebera, 2019-04-24:
I made the modification you provided and it works! I hear the shutdown melody and the device shuts down. Without that line, the melody doesn't play, and sometimes it also doesn't shut down. I have a steady 15 (14.8) V from the sensor wires.
Any ideas? Thanks for the guidance thus far.

p-h-a-i-l, 2019-04-24:
Ok, maybe no hw problem :)

How did you hardcode the speed? Try replacing
https://github.com/NiklasFauth/hoverboard-firmware-hack/blob/f06ce1fcd33c2e9238f2276858ac639fc2b03674/Src/main.c#L224
with `speed = 100;`

krebera, 2019-04-24:
Okay here are my findings:
Regardless of motor connection, shutdown melody will not play unless line 80 is `if(1){` as mentioned earlier.
Regardless of what line 80 is, changing line 224 to `speed = 100;` does not get the motors to turn.

However, I accidentally plugged in the ST-Link to my laptop before powering the board on and the motors began turning!(!!!). This was before I changed line 224 at all. The far wheel spun fast and the close wheel spun slow. I don't really know why.

Hopefully that information is helpful :)

p-h-a-i-l, 2019-04-24:
weird.. I'd first try to find out whtas causing your strange speed and steer set point. What input method are you using? The shutdown problem tells you, that speed is >20, even when you are not giving any input. Next I'd try to get serial up and running. The serial debug mode does not respond to anything, it just spits out data. Try both interfaces, expect RX and TX always to be wrong :)

krebera, 2019-04-24:
So this is embarrassing. I had a hunch this could've been the problem and neglected to post about it. I had switched out the silver power button for a switch instead, and I think that was messing around with the code. I replaced the switch back with the button, and the wheels now turn at 100! That being said, shutdown doesn't work properly when line 80 isn't `if(1){` but I think that is correct, since speed is > 20 in this situation.

I'm going to keep the button on and double back really quick to see if I can get any input / serial debug working.

Apologies if my little switch wasted your time. I greatly appreciate the help thus far.

p-h-a-i-l, 2019-04-24:
yeah, that explains some things :)
If you want to know what your controller was doing: 
https://github.com/NiklasFauth/hoverboard-firmware-hack/blob/f06ce1fcd33c2e9238f2276858ac639fc2b03674/Src/main.c#L276-L278

krebera, 2019-04-24:
Wow everything works perfectly now. ADC control, Debug, even the UART control works great.
Thanks for the help! You were very supportive and helped me trouble shoot things the right way.

Could I get the switch to work if I edited that 'if' statement? Would that be dangerous at all to do?

p-h-a-i-l, 2019-04-24:
The system is designed to operate with a push button to start and shut down. Using a switch would require lots of changes, maybe even hw. I don't really see a benefit of adding this.
Changing the if statement would mess with the shutdown.

SiChiTong, 2019-11-24:
Sirs I am also facing the same situation.
Does it mean that all we need to power on the machine with only a trigger edge to the GPIO but not a level?

samurairai, 2020-04-29:
> 
> 
> Wow everything works perfectly now. ADC control, Debug, even the UART control works great.
> Thanks for the help! You were very supportive and helped me trouble shoot things the right way.
> 
> Could I get the switch to work if I edited that 'if' statement? Would that be dangerous at all to do?

Krebera,

Can you pls post the DEBUG connection?  I cant get it to work.  Thanks.