lucysrausch/hoverboard-firmware-hack #94: Deactivate one adc

macuwe, 2019-04-24:
Hey, i'm in the process ov convertig a hoverboard into an escooter. My problem right now is that i cant turn the scooter of and i think the happens because the second motor and hall sensor is no longer attached. 
So my question is, what would be the best way to deactivate one motor side completely? Can i just uncomment the init(adc2) in the setup? 
Regards

p-h-a-i-l, 2019-04-24:
I think your problem is, that your input value is too high. During shutdown, the code checks if speed is set to lower than 20. If not, it does not switch off.
Using only one motor shouldn't be an issue.

@NiklasFauth Can we change this behavior in your codebase please? I think this is the 10th time now that people get very confused by that. (At least where I'm trying to help)
In my fork I use beeping and retrying to shut down after a few seconds. This way people know that the controller is doing something intendet. Just pulling enable and doing nothing is really confusing.

larsmm, 2019-06-23:
I added this behaviour for safety. Normally the battery cuts off in case of high load. If it cuts off on high speed you complete loose control, breaks are not working. If it switches off on field weakening the mainboard may explode.

Pull requests for a better but safe behavoir are welcome.

p-h-a-i-l, 2019-06-24:
As said, a simple indication with beeping helps a lot already. 
This change is already part of a PR I started months ago. If someone who can merge really wants this change, I could isolate that feature. A very small change..