lucysrausch/hoverboard-firmware-hack #96: Cant unloack STM.

dymek84, 2019-05-05:
always when i try unlock STM i get that message.:

Open On-Chip Debugger 0.10.0-dev (2018-09-20-12:54)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
none separate
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Info : STLINK v2 JTAG v33 API v2 SWIM v7 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.247990
Error: init mode failed (unable to connect to the target)
in procedure 'init'
in procedure 'ocd_bouncer'

i also trying using linux and other guides about hacking hoverboards, but always w/o success. 
in linux i have same message while trying unlock STM. please tell me what i do wrong.

chaz762, 2019-05-14:
so look at your chip is it an ST chip or a counterfeit? i'm having the same trouble I found the chip maker and the supposedly have their own utility and some of these new boards have the gyros built into the main board. ugg and i can't just find and old board

Jan--Henrik, 2019-05-16:
Can you please post a picture of your setup?

dymek84, 2019-05-17:
[image]
[image]
I think you asking about this. 

Today i tried to do on another mainboard ( from actual board of my son) and same results.

p-h-a-i-l, 2019-05-17:
I think you need to connect 3.3V to the programmer too.

dymek84, 2019-05-17:
no because i connect battery, and in description is to not connect 3.3v because i can kill my mb

uhi2000, 2019-06-14:
short question,
did you solve the problem with the usb dongle (3,3Volts)?

maidenone, 2019-07-07:
I have gotten this issue several times.. sometimes it complains but still manage to flash..

if openocd fails, try https://github.com/texane/stlink
it have a higher success rate, if all other fails, fall back to st link util in windows.

But in some cases i have had St link util report unlock fail as well, but a memory compare shows that it actually managed to flash..

dymek84, 2020-02-03:
i gave up long time ago. but i need it now, did some one of you can sell one motherboard to me or know somebody who can sell it with set up o want?

4ndreas, 2020-11-21:
I also have one of these blue boards and was not able to unlock the IC. I will desolder the chip and replace it with a genuine STM32.
[image]
[image]
Besides my setup works with one of the green boards just fine. Also I can access the chip with stm32cube-programmer and read the registers and so on but every time I try to upload something this happens:
[image]
I think I eased the chip since it is not beeping or doing something else.
Working setup:
[image]
This works fine so thank you for your effort.