lucysrausch/hoverboard-firmware-hack #99: error during compilation

franck24300, 2019-06-09:
Hello
I do not know anything about programming but I would like to use this program. After modifying the config.h with Visual Studio Code to use the RC PPM command and compile by typing the make command in an Ubuntu terminal window, I get a hover.hex and hover.bin file, but when I flash in the map, I can not rotate the wheels with the radio control and a PPM receiver. There is an error during compilation, maybe it's because of that? What is the cause of this error?

Src/stm32f1xx_it.c:224:5: warning: implicit declaration of function 'PPM_ISR_Callback'; did you mean 'PPM_SysTick_Callback'? [-Wimplicit-function-declaration]
     PPM_ISR_Callback();
     ^~~~~~~~~~~~~~~~
     PPM_SysTick_Callback

[image]

larsmm, 2019-06-10:
There are warnings only. This is normal... Try CONTROL_MOTOR_TEST first.

franck24300, 2019-06-10:
Yes CONTROL_MOTOR_TEST work. But not CONTROL_PPM
The difference when I compile that's it : PPM_Systick_Callback

Candas1, 2019-06-16:
What type of receiver do you have? Have you tried changing to 8 channels in config. H?

franck24300, 2019-06-16:
Yes it's good, I changed to 8 channels in the config.h and it works.

closed 2019-06-24 by franck24300