lucysrausch/hoverboard-firmware-hack #101: STM32 Utility with USB-ST link v2 no connected

uhi2000, 2019-06-13:
Hello,
I have a problem with flashing the firmware with Windows 10. I have connected the USB -ST link v2 dongle as described without 3.3v.
The STM32 ST Utility 004 writes "Can not connect to target". I really do not use the 3.3volts? "why"???
please help.... I really not use the 3.3volts?
I have connected the motherboard with an external power supply 36V.

larsmm, 2019-06-14:
You have to hold the on-button during flashing. That powers the 3.3V. Press and hold button, flash, release button after flashing.

uhi2000, 2019-06-14:
if I put the USB-Link dongle in the USB port of my PC, then I press the connected button in the STM32 Link Utility. but it always comes the error message "no target connected" of the stick but recognized in the STM32 link. What do I'm wrong ???
as long as there is no connection with the USB stick, I can not flash!!!please help me...

uhi2000, 2019-06-15:
it may be, the mainboard is looked???

uhi2000, 2019-06-15:
everything works now

closed 2019-06-15 by uhi2000