lucysrausch/hoverboard-firmware-hack #102: Break-function +power on/off with Nunshuk Button?

uhi2000, 2019-06-15:
Hallo Niklas, 
gibt es vielleicht eine Möglichkeit mit den 2 Tastern am Nunshuk z.b.eine zusätzliche Bremsfunktion oder/und eine Ein/Aus Funktion zu verwirklichen????ich bin dabei einen Stuhl für meine Oma zu bauen,für ihre Wohnung,weil sie nicht mehr so gut auf den Beinen ist  Ich habe deine Firmware-Transpotter 2.1 geflasht mit Nunshuk. Muss nur die Geschwindigkeit stark reduzieren weil zu schnell. Darum meine Frage wegen der Bremse .( sicher ist sicher) Und da der Nunshuk ja noch 2 Button hat die man vielleicht belegen kann.....eine kurze Info von dir wäre nicht schlecht....Danke im voraus.....Gruß

uhi2000, 2019-06-15:
Hello Niklas,
maybe there is a possibility with the 2 buttons on the Nunshuk zbeine additional braking function and / or an on / off function to realize ???? I'm in the process of building a chair for my grandma, for her apartment because she is not so good on my feet I flashed your firmware transpotter 2.1 with Nunshuk. Just has to reduce the speed greatly because too fast. That's why my question because of the brake. (Sure is safe) And since the Nunshuk still has 2 Button which you can perhaps prove ..... a short info from you would not be bad .... Thanks in advance .... .Greeting

uhi2000, 2019-06-16:
Hello Niklas,
maybe there is a possibility with the 2 buttons on Nunchuk z.b. an additional braking function for the caster to control? I'm about to build a chair for my grandma, for her apartment, because she is not so good on her legs anymore. I have flashed your firmware Transpotter 2.1 with the Nunchuk. Must only the max. Reduce speed because too fast. Do you think there maybe a tip. a short info from you would not be bad .... Thanks in advance ..... Greetings

uhi2000, 2019-06-16:
Hello Niklas,
maybe there is a possibility with the 2 buttons on Nunchuk  an additional braking function for the caster to control? I'm about to build a chair for my grandma, for her apartment, because she is not so good on her legs anymore. I've flashed your last firmware transpotter master with the nunchuk. Must only the max. Reduce speed greatly because too fast. Did you think there maybe a tip or for me a firmware with throttled max.Speed? a short info from you would not be bad .... Thanks in advance ..... Greetings

larsmm, 2019-06-17:
Der Nunchuk Code ist immernoch nicht 100% zuverlässig. Es kann immernoch zu Aussetzern, plötzlichem Kontrollverlust, Rucken oder Vollgas kommen.

Auch sind größere Teile des Codes ziehmlich yolo. ZB sind die Faulthandler nicht sinnvoll konfiguriert. Wenn der Chip absemmelt bremst es beispielsweise nicht. Plötzliche Resets werden nicht gut behandelt und so. Watchdogs sollten mal eingerichtet werden. Ich weiß ungefähr was zu tun ist. Ich brauche nur mal ein bisschen Zeit.

Jeder muss selber wissen welchem Risiko er sich aussetzt. Hier werden aber andere in Gefahr gebracht, die das Risiko nicht selbst einschätzen können und wo Unfälle aufgrund des Altern noch schwerwiegender sein können. Wenn du das trotzdem machen möchtest, dann begrenze die Spannung auf Werte wo das ganz nicht schnell werden kann. Mach einen fetten Not-aus in direkte Reichweite, besser noch einen Totmannschalter. Benutze einen Analogjoystick und kein i2c. Sorge dafür, dass das ganze bremst wenn 1 oder mehrere Kabel vom Joystick ab gehen.

uhi2000, 2019-06-17:
danke für deine Info ,
da hast du auf jeden recht .ich muss auf jeden Fall ein 100% sichere Sache haben .der max.Speed ist mir auch viel zu hoch ,der müsste gedrosselt werden für ältere Leute viel zu schnell.ich habe es gestern mal 
mit einem ps2 analostick ausbrobiert ist mir zu ungenau. Ich benutze die Firmwareversion-Transpotter-Hack 2.1,weiss gar nicht ob die Version jetzt das richtige für mich ist? ich würde gerne einen richtigen Joystick anschließen .da gibt es ja zu Hauf im Internet .Mein Wunsch wäre ein Hall-Joystick .....aber wenn das mit dem Nunchuk schon nicht richtig geht hat sich das zu diesen Zeitpunkt erledigt.
Auf Videos vom Niklas sieht das eigentlich sehr geschmeidig aus mit dem Nunchuk. Ich bekomme ihn morgen mit der Post und schau mir das mal an .Für den Analogen-Anschluß was wäre den die richtige Firmware???die letzte Master-Firmware???

larsmm, 2019-06-17:
Ich habe einen Nunchuk auf analog umgebaut. Das ist trotzdem ganz schön wenig Hebelweg. Für ältere Leute wäre ein größerer Joystick an der Armlehne sicherlich besser. Guck mal bei Elektrorollstühlen.

Du möchtest den Master Branch. Dann möchtest du guten Analog-Joystick-Code coden. Der der gerade im git ist funktioniert nicht. Ich habe schon Code dafür geschrieben, bin aber noch nicht zufrieden damit. Ich habe es noch nicht comitted.

uhi2000, 2019-06-17:
ich versuche die ganze Zeit den hoverboard-firmware-hack-master zu flashen,aber beim einschalten drehen die räder nur langsam vor und zurück...nix mit lenken....

uhi2000, 2019-06-17:
I try the whole time the hoverboard-firmware-hack-master to flash, but when turning the wheels turn slowly back and forth ...- nothing to steer -.... and in the transpotter firmware NG I have 3 times press the on / off button again off is. And always I'm in another mode. when I turn on ..

uhi2000, 2019-06-18:
Die beste Lösung wäre einen Hall -Joystick. ist wohl teuer aber geil---- Den hat aber so wie ich sehe noch niemand  ausprobiert!!!besser wie nunchuk auf jeden Fall....
Ich habe ihn bestellt ....!

closed 2019-06-18 by uhi2000