lucysrausch/hoverboard-firmware-hack #103: Wheels can't turn forward/Backward

Valentin2077, 2019-06-18:
Hello,

I have an issue concerning the hoverboard's wheels.
I tried using the following Inputs :

- 2 Potentiometers
- A Gametrak
- A Joystick

The motors can turn left or right, but can't move forward or backward
I am using the TranspOtter Branch.
I modified the programm so that i can control it with 2 potentiometers.
The motors have been calibrated by using a serial port of an arduino.

Any help would be appreciated

Kind Regards.

uhi2000, 2019-06-18:
I have the same problem with the firmware-transpotter Ng-hack.mit the analog stick I can only control to the left and to the right. with the Nunchuk I have no function...

larsmm, 2019-06-18:
Transpotter branch is for transpotter only. If you want analog joystick use master branch. The analog joystick code there is not working. I have working joystick code. But the driving behaviour is not nice and it needs some more testing. I have very little time at the moment. Maybe I will push it in a few weeks.

p-h-a-i-l, 2019-06-18:
But it should therefore work with a Gametrak - that's the intended input for transpotter.

Valentin2077, 2019-06-19:
Problem solved !

It was an error of cabling, and i modified the config.h file (Line 95).
The Gametrak works perfectly.

Thanks a lot for your help !

uhi2000, 2019-06-19:
hi, a question: does your gametrak work in perfect in all 4 directions ???

uhi2000, 2019-06-19:
hi lars ,
oder könntes du mir vielleicht mal deinen fertigen Hex-file mit config.h mit dem  Anschluss Gametrak rüber schicken?????

Jan--Henrik, 2019-06-20:
Hi, you can find the latest (and only) release for the TranspOtter firmware here: https://github.com/NiklasFauth/hoverboard-firmware-hack/releases
and you can find detailed build instructions here:
https://github.com/NiklasFauth/hoverboard-firmware-hack/wiki/Build-Instruction:-TranspOtter
The release is tested and works perfectly fine.

Could you please stop hijacking threads? Thank you

closed 2019-06-20 by Jan--Henrik