lucysrausch/hoverboard-firmware-hack #104: firmwareflash _hack-master error

uhi2000, 2019-06-18:
after my flash from the master hack the wheels run after always 5sec. forward and back again..is that right ???? short info would be nice ..please help...

closed 2019-06-18 by uhi2000

larsmm, 2019-06-18:
You are in Motor test mode. Look at config.h to configure your input mode.

uhi2000, 2019-06-18:
hallo Lars,
ich habe erst jetzt gelesen das ich die Firmware-hack selber programmieren muss.Habe mir gerade das Visual Studio installiert und die config.H-Datei geöffnet.Yeah.....Yeah...ok bist dahin verstanden. aber bei mir drängt die Zeit ein wenig und ich wollte ja ein fahrbaren Stuhl für meine
Oma bauen, die in 2 wochen Geburtstag hat. Hast du vielleicht eine oder mehre config.H-Dateien für mich ,wo ich vielleicht mal sehen kann welche Werte ich verändern muss um einen Nunchuk oder analogen Joystick z.b. bei der Firmware-transpotter NG oder Master zu benutzen und mir vielleicht per e-mail schicken kannst??
Das Build kann ich ja jetzt dann selber erstellen.juhu...
ich habe gerade auf youtube ein aktuelles Video mit einer RC Steuerung gesichtet und das läuft erste Sahne https://www.youtube.com/watch?v=fROINR5-P9c.
ich weiß du bist auch dabei eine Firmware zu programmieren und brauchst noch ein bisschen Zeit...aber wenn du sowas vorab für mich hast wäre super...!!!

Gruß
Tim