lucysrausch/hoverboard-firmware-hack #108: Arduino and I2C

Arkalison, 2019-06-21:
Hi,

We have a project in progress to motorize a cart to retrieve packages and it must be controlled remotely, via a remote control, or locally with the help of a 2-axis joystick.

I mainly inquire for the local part of our project, id to use a joystcik 2 axes.

I would like to know if it is possible with Arduino to control the hoverboard card with a 2-axis joystick by communicating in I2C between my arduino card and the hoverboard card.

larsmm, 2019-06-23:
Everything you asked for is possible. But you have to write code for it.

I2C is not a good idea. It is not very reliable. Maybe UART is now reliable, too. There is massive EMV which is not good für digital signals. PPM is usable as input, but you have to keep the wires VERY short. The best option is analog joystick input.

It is better to not use an external arduino. The STM32 on the hoverboard mainboards have more then enough capacity for most applications. Have a look at main.c. It is quite easy to wirte code for it.

Arkalison, 2019-06-23:
I understand perfectly but the problem is that tutoring teachers want us to use Arduino for wheel control, because we have added a system of recovery of the voltage and the distance that are displayed on a LCD of llus us we made a flashing system with an arduino relay shield.

Do you have a track on the easiest way to use arduino to control wheels? 

We have already tried to send a PPM signal with a RC catcout out to give it a voltage for the card but it did not work.